espressif/esp-idf #956: wifi: send null to keep active

nkolban, 2017-09-03:
ESP-IDF release: Current Master as of this date

While developing ESP32 apps, I switch on full ESP_LOG function to the verbose level.  I have started to notice that when I create an access point hosted by the ESP32 and connect an external station, I start to see a message being logged every second that looks as follows:

```
I (14374) wifi: pm start, type:0

I (14375) wifi: active cnt: 8
I (24375) wifi: send null to keep active
I (34376) wifi: send null to keep active
I (44377) wifi: send null to keep active
I (54379) wifi: send null to keep active
I (64380) wifi: send null to keep active
I (74381) wifi: send null to keep active
I (84382) wifi: send null to keep active
I (94383) wifi: send null to keep active
I (104384) wifi: send null to keep active
I (114385) wifi: send null to keep active
...
```

The purpose of this issue is to ask that either we switch off this message or have a better explanation of why it is continually being produced.  While it doesn't have any obvious negative effects in operation, it is *polluting* my logs.

negativekelvin, 2017-09-03:
Maybe this message could be moved from info to debug log level. But in general it is better to filter the log stream on the capture side rather than remove log generating statements from the source side.

liuzfesp, 2017-09-06:
Thanks @nkolban @negativekelvin, the log will be put into debug level.

nkolban, 2017-09-12:
Howdy @negativekelvin ,
I use the ESP-IDF tool supplied in the build system ... i.e. "make monitor".  I'm not sure that this can be easily filtered.

Spritetm, 2017-09-12:
As an additional thing: you can modify the logging of various components at run time as well... for example, ``esp_log_level_set("wifi", ESP_LOG_ERROR);`` would only show errors from the WiFi layer while still allowing other subsystems to log at the default level

closed 2017-10-11 by FayeY