espressif/esp-idf #957: SD Card Stopped working in the latest commit

llewellynnu, 2017-09-04:
Hi , 
So I have been using SD card for quite sometime now in 1 bit mode. 

The following is my Initialisation code

void MountCard(Nubot *temp) {
	ESP_LOGI(TAG, "Initializing SD card");

	sdmmc_host_t host = SDMMC_HOST_DEFAULT();
	host.flags = SDMMC_HOST_FLAG_1BIT;
	sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
	// Options for mounting the file system.
	// If format_if_mount_failed is set to true, SD card will be partitioned and formatted
	// in case when mounting fails.
	esp_vfs_fat_sdmmc_mount_config_t mount_config = { .format_if_mount_failed = true, .max_files = 10 };
	sdmmc_card_t* card;
	esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);
	if (ret != ESP_OK) {
		if (ret == ESP_FAIL) {
			ESP_LOGE(TAG, "Failed to mount file system. If you want the card to be formatted, set format_if_mount_failed = true.");
		} else {
			ESP_LOGE(TAG, "Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.", ret);
			DefaultConfig(temp);
		}
		temp->CARDOK = 0;
		return;
	}
	// Card has been initialized, print its properties
	temp->CARDOK = 1;
	sdmmc_card_print_info(stdout, card);
}

In the latest version of the IDF on master branch, This is the output I get

E (1256) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (1256) sdmmc_cmd: sdmmc_card_init: go_idle_state (1) returned 0x107
E (1256) SDCARD: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.

If I revert to commit 9fa2a9a27070d65eb72bc54891ae29935e4b1256, It working perfectly and the card is successfully mounted and I get the following output

I (266) SDCARD: Initializing SD card
Name: SA04G
Type: SDHC/SDXC
Speed: default speed
Size: 3796MB
CSD: ver=1, sector_size=512, capacity=7774208 read_bl_len=9
SCR: sd_spec=2, bus_width=5

Am I missing something?

igrr, 2017-09-04:
Hi, thank you for reporting this regression. The fix is coming up.

alexgrauer, 2017-09-06:
Hi @llewellynnu,

I would like to make the SD card work but im new to this. Can you please let me know how to roll back to that commit you said where it was working?

Thanks

llewellynnu, 2017-09-06:
@alexgrauer 
I am also net to this.

PLease look at the documentation for git revert 
basically, use the following command:
git revert <commit>

igrr, 2017-09-07:
Fixed in 51e8f439b8d405d576cc393122813214e879399b.

closed 2017-09-07 by igrr

markwj, 2017-09-25:
I'm working off the 2.1 branch (origin/release/v2.1), and seeing the same issue.

Does this bug affect 2.1, and if so can we get the fix back-ported to that?