espressif/esp-idf #958: flash read err, 1000

ilioss, 2017-09-04:
Hi,
Already read the issue #117 related to this issue.
But with the help of this I,am not able to get rid of this err,

With the help of msys32 must working ok because I can disable in the menu the "Security features and  " which i did.
When I want to run espefuse.py burn_efuse FLASH_CRYPT_CNT  in the directory where espsecure.py and esptool.py are situated I get the measage "Theofast@Theofast-PC MINGW32 ~/esp-idf/components/esptool_py/esptool
$ espsecure.py burn_efuse FLASH_CRYPT_CNT
-bash: espsecure.py: command not found
Where do I go wrong?
Make menuconfig in folder blink in esp folder.
I,am working with a esp32-vroom-32
The problem I have is that the esp wil not hold data in his nvs memory.
Any help is appreciated.
Kind regards,
Theo

benjaminaigner, 2017-09-04:
Hi,

this error is related to a missing espsecure.py script.
Two ways to resolve the problem (if this is the only problem):

- Add the path to the PATH variable: export PATH=$PATH:/home/<your user>/esp-idf/components/esptool_py
- Call the script differently: $ **./**espsecure.py burn_efuse FLASH_CRYPT_CNT

There was another issue if you have further problems: #113 

Best regards

ilioss, 2017-09-04:
Hi Benjamin,

Thanks for your reply.
To the best of my knowledge I made a script. With the following text...
export PATH="C:/msys32/home/Theo/esp-idf/components/esptool_py/esptool"
I named the script export_idf_flash_path.sh   properties mentioned Shell Script (.sh)
Unfortunately it didn,t work.

in this folder "esptool"  there are esptool,espefuse,espsecure python files among others.
I  still get the same measage   -bash: espsecure.py: command not found

I do realise it is very difficult on distance to see what is going wrong.
Where can I read the docs. which explain the making of the script? Maybe this will help.
 As far as I know f.i. installing the esp-idf do not mention making a script neither nor in the issues. So I,am a bit confused.
Dealing with this error is advisable to read the docs. in order to understand what is flash encription.

Like to hear.
Kind regards

ps. I remain looking for a methode to solve this flash read err, 1000

benjaminaigner, 2017-09-05:
Try to call the script from the `esp-idf/components/esptool_py/esptool` path with following command:

`python ./espsecure.py`

If you get something like "usage: espsecure [-h] ...."
than at least the script is working and you can continue with removing the secure boot (see other issues). 
If there is something like "Python: command not found", you have to install python in a correct way (I cannot assist you on this topic, I don't use msys/cygwin).

Best regards

ilioss, 2017-09-05:
Hi, Benjamin,

Thanks for your swift reply.
I indeed checked the above mentioned test and is posifive.
I do get the conformation that the script is running.
$ python ./espsecure.py
usage: espsecure [-h]
                 {digest_secure_bootloader,generate_signing_key,sign_data,verify   _signature,extract_public_key,digest_private_key,generate_flash_encryption_key,d   ecrypt_flash_data,encrypt_flash_data}
                 ...
espsecure: error: too few arguments

As far as I understand and know I realy lack of python program skills. Sorry by that.
Howe ever I,am using this esp32 chip in the internet radio ( which I like very much ) with the err it is not possible to read the SD card most probably caused by the flash err.
So I now will go on further with your proposal for cont. with removing the secure boot........
Thanks
Keep my result posted.
Kind regards,
Theo

ps. my guide is the espressif idf programming guide
What is the diff between  msys2 and mingw32

ilioss, 2017-09-05:
Hi,

After further tests and RTFM I experienced that when I use "" python ./espefuse.py --port COM14 summary
It connects to the esp32 and give this as output...
Theofast@Theofast-PC MINGW32 ~/esp-idf/components/esptool_py/esptool
$ python ./espefuse.py --port COM14 summary
espefuse.py v2.1
Connecting........_
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 0 R/W (0x0)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    MAC Address                                       = 22:00:cc:00:cc:00 R/W
CHIP_VERSION           Chip version                                      = 0 R/W (0x0)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

Which looks to me all ok. all zero,s   " 0 "
Remains the err. flash read err, 1000 after 2nd start of the esp32?????
Keep surging.
Regards,
Theo

igrr, 2017-09-07:
If you have `flash read err, 1000` error continuously after reset, please check that you don't have any external signals connected to SPI flash pins of the ESP32 (pins 6-11). If you don't have anything connected to these pins, it means that your SPI flash chip may have been damaged.

ilioss, 2017-09-07:
Hi, @igrr 
Ok, I will check.
Question if I made a encrypt summary I read all figures 0 "zero" expect the mac.adres from the chip
is this correct?
Kind regards,
Theo
ps. above reply from Benjamin realy helped my to find my way

igrr, 2017-09-07:
MAC address value in espefuse.py summary looks incorrect. Unless you have modified the output for presentation purposes, MAC address of an ESP32 should start with one of prefixes listed [here](https://ip.rst.im/oui/Espressif%20Inc.).

projectgus, 2017-09-08:
In addition to what @igrr says, because FLASH_CRYPT_CNT and ABS_DONE_0 efuses are both 0 this suggests that secure boot and flash encryption were never enabled and are not currently enabled. Unless you turned these options on in menuconfig and manually & explicitly flashed a bootloader, that's the case (they don't accidentally get turned on.)

For this reason, I do not recommend trying to burn any flash encryption efuses in this case, as you will enable (rather than disable) flash encryption!

If you can explain some more about your hardware setup, etc. then we may be able to help you diagnose the underlying problem. (As @igrr says, the MAC value looks very unusual and incorrect.)

ilioss, 2017-09-08:
Hi,
@iggr  the mac adres I did myself . 
@projectgus  Also many thanks for reply.
I made a drawing howe I connect the radio/esp32 
This is the ESP32 internet radio from ED Zelf  ( very nice project )
Enclosed you will find the drawing
[image]
From a powersupply 230/50Hz sec. 6VDC 400mA ( regilated I feeding a 3.3 volt regulator on the 3.3 volt side about 1160 micro Farad elco capacitors. I have the ideer the power supply is ok.
Draw about 250 milli amps@3,3 volt
Indeed I hesistate to burn efuses.
Because the summary showed all 0  zero,s  which must be ok.
The mac adres is changed by myself. ( security)
I suffer from the "flash read err, 1000 " which after upload is not present only after I started the radio by the on/off switch the error ocure.
Which on his turn do not tart the SD card.
This ESP32 devboard I played with several scketches for gaining some experience. Like simple server. etc.
A second radio with a similair setup and  out of the box ESP32 give no problem.
 So either a HW or Sw fault I,am looking for.
Software is most likely out of suspection.
HW bad powersupply or firmware setting corrupt must be the problem.
Like to hear. 
Theo

[image]

ilioss, 2017-09-09:
Hi,
In addition to former meassage I enclose debug list from start-up esp32
first start after update of the software from the default page (NVS)
second start after start  by means of push the EN button on devkit board

First the SD card is accessible.
Second start SD card not accessible and flash read err, 1000 error.
[ESP32 reset met html page config.txt](https://github.com/espressif/esp-idf/files/1289666/ESP32.reset.met.html.page.config.txt)
[ESP32 opstart na reset with EN.txt](https://github.com/espressif/esp-idf/files/1289668/ESP32.opstart.na.reset.with.EN.txt)

negativekelvin, 2017-09-09:
https://github.com/espressif/esp-idf/issues/113#issuecomment-268698113
https://github.com/espressif/esp-idf/issues/113#issuecomment-303186904

Both your logs say 

> D: Locate mp3 files on SD, may take a while...
D: 11 tracks on SD

ilioss, 2017-09-09:
Hi, @-negativekelvin
Cool to notice.
Apart from this sd card can`t be read with the flash read err,1000 after startup with poweron.
I will make a new printout.
So # 113 deals also with this flash read err. I will read carefully.
Btw I,am redesighn my pcb to by shure to work with a spoton powersupply.
Kind regards,
Theo

projectgus, 2017-09-11:
Are you able to remove the DEVKIT V1 board from the rest of the hardware entirely, and see if it programs & boots successfully connected with only USB connected?

ilioss, 2017-09-11:
Hi, all
In reaction of your all reply,s I report as follow, after exchange my esp32 devboard it is defenetely a hardware fault in my pcb.
I changed a "good" esp32 into my faulty radio and the fault remains.
I inserted my "faulty" esp32 into my good radio and the fault was not noticed.
So I have a seriuos hardware fault in my radio.
Last weekend I already cleared up my wiring and checked the pcb for faults visual no adnormalities found.
So I have a job to do.
Hopefully this takes away some smoke around the flash read err, 1000 error.
Keep you posted.
All I do realy appreciate the contribution in reply on my post.
Kind regards,
Theo   
Again I like to mention the wonderfull project EdZelf has made where I use my ESP32 
@projectgus yes I am able to just single test my both ESP32 and they where possible to program.
@-negativekelvin again I tested the printout and the same result however the flash fault and SD card was read It is not possible from http and the display mentioned sd card mount failed.

ilioss, 2017-09-11:
Hi,

After  further testing and fault finding I concluded that a micro SD card from 1G is the faulty part.
I placed a new 4G micro SD card in the SD adapter and tested both radio,s correct functioning.
I noticed during testing that my ceramic capaciters in my powersupply do there work.
Before my LD33CV I soldered a 470nF and after a 100nF capaciter. Glitsing which happened plugging the USB plug is not anylonger noticed.
All of you who contribute to this issue many thanks.
Kind regards,
Theo
I closed this issue

closed 2017-09-11 by ilioss

kbahey, 2018-06-30:
Although this is a solved issue, it is worth mentioning here another variant, for those coming here via Google.

If you plug your ESP32 into a board that you made, and you see the 'flash read err, 1000', but it works fine when you unplug it from your board, then here is a case which would trigger this:

Basically, if you have a tiny solder bridge between GND and one of the FLASH pins (GPIO6, 7, 8, 9, 10, 11), this will happen. 

To solve it inspect your board with a magnifying glass carefully.