espressif/esp-idf #961: OTA example getting 'stuck' on loading subtype 0x11 (OTA App #2)

jnorgan, 2017-09-05:
I was using the OTA example code and after two OTAs, it was getting stuck loading only the second OTA app (subtype 0x11). It looks like the sequence value for A was incremented to 0x00000003 and since the bootloader is using the following:

https://github.com/espressif/esp-idf/blob/9da1bf1a45f670a2d14083a9865f10826a7bca43/components/bootloader/subproject/main/bootloader_start.c#L332

`get_selected_boot_partition()` continues to return `2` and the bootloader loads OTA App #2 instead of cycling back to load the first slot. 

I haven't dug into the ota_ops code enough to contribute a proper fix. I'm doing a hacky `return (MAX(sa.ota_seq, sb.ota_seq) - 1) % 2;` in the above bootloader code, but this can't be the correct fix for this. 

Anyone able to help on this?

negativekelvin, 2017-09-06:
https://github.com/espressif/esp-idf/pull/955

jnorgan, 2017-09-06:
👍 Yep. That's exactly it. Sorry for the duplicate.

closed 2017-09-06 by jnorgan