espressif/esp-idf #963: Commit 4776f73c breaks SPI DMA

nickvanderstel, 2017-09-06:
Commit 4776f73ca4651a5bb65b4db2407205c0dd39925e disables peripheral clocks by default, including the SPI DMA clock. This clock is not enabled when periph_module_enable() is called so SPI transfers with DMA enabled will always fail.

Is this DMA shared between all the SPI modules? Maybe this clock should be left running by default...

jolivepetrus, 2017-09-07:
From the Lua RTOS team I can confirm this issue. Nothing in Lua RTOS works with SPI. We will fixed this by disabling DMA transfers in our SPI driver in a few minutes. Disabling DMA all works well.

igrr, 2017-09-07:
Hi, the clock disabling commit broke SPI DMA, SDMMC, Ethernet, SDIO slave, and possibly the third-party CAN driver. We have merged the fix for this a couple of days ago, but due to some issues the master branch on Github has not been synchronized yet. Please use the attached patch set for now.

[clock-enable-patches.zip](https://github.com/espressif/esp-idf/files/1283250/clock-enable-patches.zip)

jolivepetrus, 2017-09-07:
We can confirm that CAN is not affected.

igrr, 2017-09-07:
Fixed in 51e8f439b8d405d576cc393122813214e879399b.

closed 2017-09-07 by igrr