espressif/esp-idf #964: TG0WDT_SYS_RESET error after POWERON_RESET / DEEPSLEEP_RESET

clempar, 2017-09-06:
Hi,

Recently, after a esp-idf update, I have noticed that my test projects using deep standby didn't work correctly. I couldn't get the expected wake-up reason. 

Looking at the logs, I saw that immediately after every reset (POWERON_RESET, DEEPSLEEP_RESET) I have a second "TG0WDT_SYS_RESET" reset (sometimes two). A watchdog resets the chip a second time and as a result I loose the initial wake-up reason (as well as 1 or 2 s of boot time).

I have reproduced the issue with system/deep_sleep example:
```
[First boot]
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[...]
Not a deep sleep reset
Enabling timer wakeup, 20s
Enabling EXT1 wakeup on pins GPIO25, GPIO26
Touch pad #8 average: 625, wakeup threshold set to 525.
Touch pad #9 average: 643, wakeup threshold set to 543.
Enabling touch pad wakeup
Enabling ULP wakeup
Entering deep sleep
[20s later]
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
[...]
Not a deep sleep reset
[...]
```
Full logs attached (first boot and first wake-up).
[minicom.txt](https://github.com/espressif/esp-idf/files/1282809/minicom.txt)

Using git bisect I traced the issue back to commit 123966deeaf0e60944c697ef8f529bcd48b1e317 " Also do dport stalling in esp32 component" . 
I confirm that with the previous commit I have normal wake-up, and that this commit introduces TG0WDT_SYS_RESET issues.

I tried to revert the change introduced by this commit on top of master ( commit  9da1bf1a at the time of writing) and it appears to fix the issue, at least on deep_sleep example. 

Can you reproduce the issue ? If it makes any difference my board has a Rev0 ESP32, and don't have a Rev1 to test with. I don't think this is the known reset problem of rev0 silicon, because it used to work OK and commit 123966d  introduced a regression.

I don't understand exactly what is the link between the change and the comment of this commit, could you take a look and make sure there is no mistake ? 

I can provide some logs or do some tests if needed.
Thank you.

igrr, 2017-09-07:
Hi, we have merged the fix for this issue a couple of days ago, but due to some issues master branch on Github can not be synchronized at the moment. You can use the attached patch in the meantime.

```patch
From b9ff771e2c0b4cc185acf809127a6ad38001d706 Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Fri, 1 Sep 2017 18:35:42 +0800
Subject: [PATCH] deep sleep: don't use protected DPORT access in wake stub

---
 components/esp32/sleep_modes.c | 6 ++++--
 1 file changed, 4 insertions(+), 2 deletions(-)

diff --git a/components/esp32/sleep_modes.c b/components/esp32/sleep_modes.c
index 9cf88672..e305b96e 100644
--- a/components/esp32/sleep_modes.c
+++ b/components/esp32/sleep_modes.c
@@ -100,8 +100,10 @@ void esp_set_deep_sleep_wake_stub(esp_deep_sleep_wake_stub_fn_t new_stub)
 
 void RTC_IRAM_ATTR esp_default_wake_deep_sleep(void) {
     /* Clear MMU for CPU 0 */
-    DPORT_REG_SET_BIT(DPORT_PRO_CACHE_CTRL1_REG, DPORT_PRO_CACHE_MMU_IA_CLR);
-    DPORT_REG_CLR_BIT(DPORT_PRO_CACHE_CTRL1_REG, DPORT_PRO_CACHE_MMU_IA_CLR);
+    _DPORT_REG_WRITE(DPORT_PRO_CACHE_CTRL1_REG,
+            _DPORT_REG_READ(DPORT_PRO_CACHE_CTRL1_REG) | DPORT_PRO_CACHE_MMU_IA_CLR);
+    _DPORT_REG_WRITE(DPORT_PRO_CACHE_CTRL1_REG,
+            _DPORT_REG_READ(DPORT_PRO_CACHE_CTRL1_REG) & (~DPORT_PRO_CACHE_MMU_IA_CLR));
 #if CONFIG_ESP32_DEEP_SLEEP_WAKEUP_DELAY > 0
     // ROM code has not started yet, so we need to set delay factor
     // used by ets_delay_us first.
-- 
2.11.0 (Apple Git-81)

```

igrr, 2017-09-07:
Fixed in c007213fedc0881c99bbe5657bc51cb3869c00c9.

closed 2017-09-07 by igrr

Tasshack, 2017-10-01:
Hi,
This problem is not fixed yet. When it happens, wdt resets cpu before even wake stub has triggered. You can not reproduce this after flashing the chip but you have to power it down and power up again to reproduce the issue. I think wake stub is not the right place for clearing the DPORT_PRO_CACHE_CTRL1_REG bit.
I also tried with 5000us wake delay but wdt reset is still happening.