espressif/esp-idf #965: Cant get a MicroSD card to work.

alexgrauer, 2017-09-07:
Hi all,

Im trying to make a microSD card to work but im not being able.

I have the following boards:

ESP-WROOM-32: http://www.ebay.com.au/itm/262953936783
Micro SD shield #1: http://www.ebay.com.au/itm/191736124794
Micro SD shield #2: https://www.amazon.com/gp/product/B071WJ99N8/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1

If I use the Micro SD shield #1 I don't put the resistors defined in the SD Card Example because it seems as it already has them.
If I use the Micro SD shield #2 then I do everything as it says in the example.

But for sure im doing something wrong.
Im not being able to make it work in any of the modes 4-line, 1-line or SPI.

When I use Micro SD shield #1, I do the following connections for SPI:
IO13 -> CS
IO15 -> MOSI
IO2 -> MISO
IO14 -> CLK
GND -> GND
3.3V -> 3.3V (I also tried with the 5V)

but I always get:
I (0) cpu_start: App cpu up.
I (577) heap_init: Initializing. RAM available for dynamic allocation:
I (597) heap_init: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (616) heap_init: At 3FFB2748 len 0002D8B8 (182 KiB): DRAM
I (635) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (655) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (674) heap_init: At 400880F4 len 00017F0C (95 KiB): IRAM
I (693) cpu_start: Pro cpu start user code
I (752) cpu_start: Starting scheduler on PRO CPU.
I (191) cpu_start: Starting scheduler on APP CPU.
I (191) example: Initializing SD card
I (191) example: Using SPI peripheral
I (191) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1241) sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107
E (1241) example: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.

When I use Micro SD shield #2, I do the connections defined in the example.
What can be wrong? Maybe I don't understand something about GPIO12? wat should I do with it, I mean, about all it says in the Note, I don't get it 100%.

I used two different micro sd cards, but nothing.

Anyone with any ideas?
Thanks in advance

Alex

igrr, 2017-09-07:
Could you please mention the commit id of ESP-IDF?

If this is a recent one, this is a possible duplicate of #957 (in which case the patch is provided in #963).

alexgrauer, 2017-09-07:
Well im quite new to all this so im not sure if im working correctly with GitHub, but I created to different folders: esp and esp2
when I run git log in esp, I get:

commit 2861f3e88ee4f08913c8ae77195d094412c9247d
Merge: 6f52cf0f a01db556
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Tue Aug 22 23:39:53 2017 +0800

than I run it in esp2, I get:

commit 9fa2a9a27070d65eb72bc54891ae29935e4b1256
Author: He Yin Ling <heyinling@espressif.com>
Date:   Tue Aug 29 23:09:02 2017 +0800

    CI: remove build at from CI:
    
    AT is a standalone application. It will not keep up with latest IDF. If
    IDF changes APIs which AT used, CI will fail. Therefore remove AT job.

I understand the in esp2 is the one in which it was working fine, right?

igrr, 2017-09-07:
Please try updating to the latest master branch. The issue should be fixed now.

FayeY, 2017-09-11:
Hi @alexgrauer , have you tried the latest IDF? Is this problem solved now?

jolivepetrus, 2017-09-12:
In Lua RTOS we receive this errors:

E (5257) sdspi_host: data CRC failed, got=0x98d3 expected=0xcbe3
E (5259) sdspi_host: sdspi_host_start_command: cmd=17 error=0x109
E (5260) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (5267) diskio_sdmmc: sdmmc_read_blocks failed (265)

igrr, 2017-09-12:
@jolivepetrus, can you give more information about your hardware setup (i.e. circuit diagram)? Does the error also happen if you add `host.max_freq_khz = SDMMC_FREQ_PROBING;` line to SDMMC host initialization? 

Based on the log, SDMMC peripheral has been initialized correctly, and the card probing was successful, but there was a CRC mismatch during data transfer between the peripheral and the card.

jolivepetrus, 2017-09-12:
If adding host.max_freq_khz = SDMMC_FREQ_PROBING, works well.
Setting host.max_freq_khz = 19000 works well
Setting host.max_freq_khz >= 20000 fails

It seems that fail with SDSC cards.
With an SDHC/SDXS cards seems to work well without any changes in max_freq_khz.

Maybe the idf driver doesn't check for SD Card max speed?

Connections:

MISO=GPIO19
MOSI=GPIO23
CLK=GPIO18
CS=GPIO26

Attached you have the schematic.

[image]

igrr, 2017-09-12:
SD card max speed (reported by the card) is always 25MHz, until switch_func command is executed to enable high-speed mode. The behavior here is the same for SDSC and SDHC. If you set max_freq_khz parameter to SDMMC_FREQ_DEFAULT, then the driver will use 20MHz frequency and will not attempt to put the card into high-speed mode.

The question is, did the IDF driver work with the same cards and the same hardware previously? If it did, then the issue is likely related to SDMMC peripheral clock changes which were implemented to support high-speed mode.

Another question, why does your schematic mention SPI mode signals (CS/MOSI/MISO/CLK) when you are in fact using the card in SD mode?

jolivepetrus, 2017-09-12:
@igrr,

In my test hardware I'm using an SD to micro SD adapter.

[image]

igrr, 2017-09-12:
Ok, so then CS (DAT3) pin is not connected to the ESP32, is that correct?

Also, please answer my other question above (whether same hardware worked with previous IDF version).

jolivepetrus, 2017-09-12:
For clarify.
The schematic for microSD is:

[image]

Somewhere in the time the same hardware worked, but I'm realised that my tests are done with an unbranded SD Card (2 Gb microSD). It's difficult for me to find the right combination of idf / Lua RTOS commits for demonstrate it.

With a Kingston 16Gb HC, works.

igrr, 2017-09-12:
Okay, in that case i am going to close this issue. If you happen to find a case when same hardware used to work but doesn't work with the current version of IDF, please feel free to reopen.

We plan to add a feature to SDMMC driver to do automatic tuning of read/write clock edges, which may improve robustness at high speeds.

closed 2017-09-12 by igrr

jolivepetrus, 2017-09-12:
Somewhere in the time the same hardware worked using Lua RTOS SD driver. In that driver some SD cards where configured at 12.5 Mhz.

	/* Switch to the high speed mode, if possible. */
	if (u->csd[4] & 0x40) {
		/* Class 10 card: switch to high-speed mode.
		 * SPI interface of pic32 allows up to 25MHz clock rate. */
		card_high_speed(unit);
	}

https://www.sdcard.org/developers/overview/bus_speed

igrr, 2017-09-14:
You aren't confusing 12.5 MHz and 12.5 MByte/s, are you?

In default speed mode (25MHz / 12.5MByte/s), which should be supported by all SD cards, we use 20MHz clock frequency, because the clock divider can only produce 160MHz/n, i.e. either 20 or 26.7 MHz. Support for high speed mode (50MHz / 25MByte/s) is checked at runtime, and if the card supports it, SDMMC driver will switch it into high speed mode. For high speed mode, we use 40 MHz clock frequency (due to a limitation of clock divider, only 53.3MHz or 40MHz frequencies are possible).

jolivepetrus, 2017-09-20:
Yes, I'm confused.
I will destroy this strange SD Card.

N-Coder, 2018-11-18:
I've got the same issue here, a lot of mostly 0 "cmd response"s and then suddenly a `ESP_ERR_TIMEOUT` / 0x107 with the SD card example in SPI mode (no other changes, except for enabling `#define USE_SPI_MODE`). As soon as I set `host.max_freq_khz` to anything below 20000 (i.e. 19000 or `SDMMC_FREQ_PROBING`) it works flawlessly:
```
Name: SC32G
Type: SDHC/SDXC
Speed: 19 MHz
Size: 30436MB
```

My esp-idf is on commit fb7ba1baa068a2267ef54b7105cf3c8a004fe4eb and I'm using a [Catalex MicroSD Adapter](http://www.vishnumaiea.in/projects/hardware/interfacing-catalex-micro-sd-card-module) with a [Sandisk Ultra 32GB MicroSDXC](https://www.amazon.de/gp/product/B073JWXGNT).

chipweinberger, 2021-04-27:
Solved this issue!

I was hitting 0x107 as well and it turns out I just was not supplying enough 3.3v power to VDD.

I had my SD card attached to a GPIO and using that as the 3.3v supply, but GPIOs can only do 40mA and SD cards can peak about 100mA so it wasn't enough. Giving it a dedicated 3.3v power rail worked.

knee-cola, 2021-12-07:
I'm also getting 0x107 error with 32GB MicroSD card (FAT32) while running the latest IDF SD card example on Titigo Lora32 LED board (with integrated SD card slot).

Here's the log I'm getting:
```
I (310) example: Initializing SD card
I (320) example: EXAMPLE_FORMAT_IF_MOUNT_FAILED = TRUE
I (320) example: Using SPI peripheral
I (330) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (380) sdspi_transaction: cmd=5, R1 response: command not supported
E (1390) sdmmc_common: sdmmc_init_cid: send_cid returned 0x107
E (1390) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
```
The example **works correctly** on an old 2GB FAT16 card.
This shows that the card slot is **connected correctly** to the board. 

Now I don't know if this is caused by hardware (32GB card using too much power) or software (library not supporting FAT32 or an error in the given example).

stefan123t, 2022-10-29:
Just a reminder that the problem may be caused by a hardware problem too
and #965 seems to be the mother of all such issues here in esp-idf. 

```c
sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
[E][SD_MMC.cpp:85] begin(): Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.
```

Here Jason checked a lot of 20 ESP32CAMs and 6 had problems with the MicroSD Card. 

ESP-32-cam “Mounting SD card” Error
https://rntlab.com/question/esp-32-cam-mounting-sd-card-error/
> I have 20 esp32-cams and 14 of them work perfectly, but 6 are all giving me the same error.
> I have set each one up identically

That is a 30% chance that a modules SD Card is not working!
My chance with 25% is a bit lower with 1 out of 4 not working.
But trying hard to get that broken one working first is annoying ;)

Thanks for your great software tools!