espressif/esp-idf #968: ESP32] Change the core running Wifi Task and Bluetooth Task

hiik3n, 2017-09-07:
I use ESP32 to scan ble data and transmit scanned data to the internet via wifi connection.
I found that the number of scanned ble data when only bluetooth is active is larger when both bluetooth and wifi are active.
does anysone experience on that?
Both wifi task and bluetooth bluetooth task run on core 0, is it possible to change blue task and wifi task run on prefered core? Will it increase the number of scan ble data and how to do it?

Thank you

negativekelvin, 2017-09-07:
In esp32 WiFi and Bluetooth share the same radio and have to cooperate so that sometimes WiFi is active and sometimes Bluetooth is active.  Other Bluetooth devices do not know about WiFi so they may advertise during wifi active time.

hiik3n, 2017-09-08:
@negativekelvin Thank you for your answer, you have any idea to control when wifi run and when bluetooth run. Because when I active wifi, sometime I can not scan any ble package.

hiik3n, 2017-09-11:
@TianHao-Espressif after check the task priority:
main           	R	1	1000	4
IDLE           	R	0	556	6
IDLE           	R	0	556	5
Btc_task       	B	19	8076	38
tiT            	B	18	1548	11
Tmr Svc        	B	1	1380	7
hciH4T         	B	21	1848	40
btuT           	B	20	3028	41
btController   	B	23	2124	43
hciHostT       	B	22	1828	39
ipc1           	B	24	544	2
bt_control_task	B	9	1556	16
pmT            	B	21	1004	14
wifi           	B	23	1964	13
rtT            	B	22	2988	15
eventTask      	B	20	2436	12
ipc0           	B	24	632	1

If I try set priority of bt task (hci, controller,..), will the ble scan performance be improved?

TianHao-Espressif, 2018-01-05:
IDF release/v3.0 could change which core running by  bluetooth controller and host.

liuzfesp, 2018-03-05:
Currently, WiFi can only run on core#0

closed 2018-05-18 by igrr

gangalko98, 2020-10-28:
Can anybody provide code for running classic Bluetooth and Wi-Fi together, running through different cores??