espressif/esp-idf #971: Malloc failure issue

lucazader, 2017-09-08:
Hi all,

I am having a very strange issue at the moment.

I am having the esp32 crash during either malloc or a free.
Sometimes the crash occurs when calling malloc from my application code (trying to allocate a buffer of between 28 and 100 bytes).
Other times it seems to happen from within the BLE stack.
There is plenty of heap left (~149k)
I have tried on both v2.1(current head) and v3.0 (commit 2861f3e8)

This seems to happen after I receive a few BLE messages and then try to send these out over the uart (at least on v2.1)

Here is the backtrace from the crash on v3.0 of the idf:
```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40087d7e  PS      : 0x00060b33  A0      : 0x800822d5  A1      : 0x3ffd74a0
0x40087d7e: multi_heap_malloc at C:/0000-Working/ESP32/esp-idf/components/heap/multi_heap.c:547 (discriminator 1)

A2      : 0x3ffcf590  A3      : 0x0000004c  A4      : 0x3ffdafa0  A5      : 0xffffffff
A6      : 0x3ffdaf3c  A7      : 0x3ffdc1ac  A8      : 0x0000007c  A9      : 0x3ffd7480
A10     : 0x000002b0  A11     : 0x00000003  A12     : 0x00060923  A13     : 0x3ffc1b88
A14     : 0x00000004  A15     : 0x3ffdc1bc  SAR     : 0x00000008  EXCCAUSE: 0x0000001c
EXCVADDR: 0xffffffff  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

Backtrace: 0x40087d7e:0x3ffd74a0 0x400822d2:0x3ffd74d0 0x4008686d:0x3ffd74f0 0x400869f0:0x3ffd7510 0x40086e2c:0x3ffd7530 0x
400826b2:0x3ffd7550 0x400826d4:0x3ffd7570 0x400827dd:0x3ffd75b0 0x4000be35:0x3ffd75d0 0x4005789a:0x3ffd75f0 0x40056f25:0x3f
fd7680 0x401039bd:0x3ffd76d0 0x40087aed:0x3ffd7700 0x40087eb1:0x3ffd7720 0x40082335:0x3ffd7740 0x40082559:0x3ffd7760 0x4000
bec7:0x3ffd7780 0x401120e5:0x3ffd77a0 0x40114065:0x3ffd77c0
0x40087d7e: multi_heap_malloc at C:/0000-Working/ESP32/esp-idf/components/heap/multi_heap.c:547 (discriminator 1)

0x400822d2: heap_caps_malloc at C:/0000-Working/ESP32/esp-idf/components/heap/heap_caps.c:123

0x4008686d: pvPortMalloc at C:/0000-Working/ESP32/esp-idf/components/freertos/port.c:438

0x400869f0: xQueueGenericCreate at C:/0000-Working/ESP32/esp-idf/components/freertos/queue.c:1992

0x40086e2c: xQueueCreateMutex at C:/0000-Working/ESP32/esp-idf/components/freertos/queue.c:495

0x400826b2: lock_init_generic at C:/0000-Working/ESP32/esp-idf/components/newlib/locks.c:79

0x400826d4: lock_acquire_generic at C:/0000-Working/ESP32/esp-idf/components/newlib/locks.c:132

0x400827dd: _lock_acquire_recursive at C:/0000-Working/ESP32/esp-idf/components/newlib/locks.c:169

0x401039bd: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/
../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)

0x40087aed: get_next_block at C:/0000-Working/ESP32/esp-idf/components/heap/multi_heap.c:547 (discriminator 1)

0x40087eb1: multi_heap_free at C:/0000-Working/ESP32/esp-idf/components/heap/multi_heap.c:547 (discriminator 1)

0x40082335: heap_caps_free at C:/0000-Working/ESP32/esp-idf/components/heap/heap_caps.c:172

0x40082559: _free_r at C:/0000-Working/ESP32/esp-idf/components/newlib/syscalls.c:33

0x401120e5: GKI_freebuf at C:/0000-Working/ESP32/esp-idf/components/bt/bluedroid/gki/gki_buffer.c:411

0x40114065: btc_task at C:/0000-Working/ESP32/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:91
```

I am really not sure what is going on here. If anyone can help me out it would be greatly appreciated.
Let me know if you need anything else, eg more specific information about what the system is doing before the crash etc.

**Some of the things i have looked into:**
- main idle task seems to still be running
- free heap is ~149k
- the user code malloc doesnt seem to crash until after i connect as a central to two peripherals and receive a message or two from both peripherals.

projectgus, 2017-09-08:
Hi @lucazader,

The underlying cause is some kind of heap corruption.

To decode the stack trace you posted, going from bottom (outside) to top (in):

* Bluedroid code goes to free some buffer.
* Inside the free function multi_heap_free() there's an inline function get_next_block(), which has an assertion `assert(next > (intptr_t)block);`. If this assertion fails, the heap is corrupted (something else has written over the heap's internal metadata structure.)
* When the assertion fails it tries to print the assertion error message, which is vai printf().
* The first time you printf() in a task it tries to allocate a lock.
* Allocating the lock calls malloc() and malloc() is exploding (again) because the heap is corrupt.

But the underlying cause is that something, somewhere, prior to this code executing has corrupted the heap.

We have some new tools for tracking heap corruption coming in the next few days to IDF. This should help you track down the problem - but for now I'd suggest looking in your code for anything which might overrun when writing a buffer, or dereference a pointer incorrectly. Unfortunately because of the nature of this crash, the problem could be anywhere.

projectgus, 2017-09-08:
(Note the bug doesn't seem to be a double-free or something else directly related to the buffer that bluedroid is freeing at the time of the crash. There are a number of assertions in multi_heap_free() to check for that kind of situation and these all pass. The failure is when multi_heap_free() goes to walk the heap looking for neighbouring blocks to the one which is being freed.)

ESP32DE, 2017-09-08:
@lucazader 
not sure is this a soltion for you, "too"
but have a try with this "custom" coment on 
"freertos/heap_regions.c
"vPortFreeTagged"
[read details here](https://esp32.com/viewtopic.php?f=2&t=2713&p=13463#p13463)

> 
> ```
> /* Check the block is actually allocated. */
>         configASSERT( ( pxLink->xAllocated ) != 0 );
>         // configASSERT( pxLink->pxNextFreeBlock == NULL );  // note: customized by rudi so please wait for the ESP-IDF 3.0 what they do :)  
> 
> 
> ```

give feedback how you get on in this.

best wishes
rudi ;-)

lucazader, 2017-09-10:
Thanks for your responses.

It was indeed heap corruption from a buffer over-run.

Thanks for all your help!

closed 2017-09-10 by lucazader