espressif/esp-idf #972: Any time-based job schedulers available in ESP-IDF?

huming2207, 2017-09-08:
Hi all,

I would like to ask that is there any time-based job timer libraries available (similar to cron in Linux) in ESP-IDF and/or FreeRTOS? Or is there any good idea for implementing these functions?

By the way, sorry if I asked dumb question...

Thanks and regards,
Jackson

DVSUHZ-g, 2017-09-08:
Also curious.

Spritetm, 2017-09-12:
It depends on what you want exactly... you could say the FreeRTOS timer API can be abused as a barebones cron replacement, for example. If you want anything more complex, you may have to build it yourself (or see if someone else already did).

closed 2017-09-21 by huming2207

huming2207, 2017-09-21:
@Spritetm Thanks for your reply. Now I will plan making a wrapper library for the cron stuff.