espressif/esp-idf #974: Station Mode Connection Error (bcn_timout,ap_probe_send_start)

santanurahaman, 2017-09-08:
Hi,

I have a simple application to connect ESP32 to a Wifi Network. 
After it's connected successfully to the network, this log is kept coming continuously  `I (50647) wifi: bcn_timout,ap_probe_send_start` and finally it's getting disconnected from the Wifi Network in few minutes. I am facing this issue with the recent update of the esp-idf.
Does anyone have any idea what is causing this issue? Thanks.

Log: 
```
I (785) wifi: Init dynamic tx buffer num: 32
I (785) wifi: Init data frame dynamic rx buffer num: 32
I (790) wifi: Init management frame dynamic rx buffer num: 32
I (796) wifi: wifi driver task: 3ffbd268, prio:23, stack:4096
I (801) wifi: Init static rx buffer num: 10
I (805) wifi: Init dynamic rx buffer num: 32
I (809) wifi: Init rx ampdu len mblock:7
I (812) wifi: Init lldesc rx ampdu entry mblock:4
I (817) wifi: wifi power manager task: 0x3ffc26e0 prio: 21 stack: 2560
I (844) phy: phy_version: 359.0, e79c19d, Aug 31 2017, 17:06:07, 0, 0
I (845) wifi: mode : sta (24:0a:c4:0c:ba:28)
I (846) WIFI: Event Ocured: 2
I (970) wifi: n:10 2, o:1 0, ap:255 255, sta:10 2, prof:10
I (1627) wifi: state: init -> auth (b0)
I (1648) wifi: state: auth -> assoc (0)
I (1657) wifi: state: assoc -> run (10)
I (1706) wifi: connected with TestWifiNetwork, channel 10
I (1709) WIFI: Event Ocured: 4
I (3837) event: ip: 192.168.0.7, mask: 255.255.255.0, gw: 192.168.0.1
I (3837) WIFI: Event Ocured: 7
I (4657) wifi: pm start, type:0

I (50647) wifi: bcn_timout,ap_probe_send_start
I (51148) wifi: bcn_timout,ap_probe_send_start
I (51648) wifi: bcn_timout,ap_probe_send_start
I (59966) wifi: bcn_timout,ap_probe_send_start
I (60467) wifi: bcn_timout,ap_probe_send_start
I (60968) wifi: bcn_timout,ap_probe_send_start
I (61468) wifi: bcn_timout,ap_probe_send_start
............
...........
I (201669) wifi: state: run -> auth (7c0)
I (201669) wifi: n:10 0, o:10 2, ap:255 255, sta:10 2, prof:10
I (201670) wifi: pm stop, total sleep time: 0/197012403
```

xarrange, 2017-09-30:
Same to me

shirish47, 2017-10-30:
anybody resolved this ?

kakegu, 2017-11-03:
Same to me

paulreimer, 2017-12-08:
This has just started happening to me as well, on what I had considered good hardware/software. My OTA server had changed their root CA, so I added it to my root cert file, otherwise no changes. And I cannot do complete OTA session any more.

paulreimer, 2017-12-08:
I did a short power cycle of my dev board and it still didn't work. Then I did a full rebuild (`rm -rf build && make erase_flash flash monitor`) and a long power cycle, and it seemed to work. I should note this was transient behaviour -- I didn't change my code between builds.

jack0c, 2018-01-03:
@paulreimer Seems that you have erase the flash include NVS. Maybe you erase the phy calibration data, so ESP32 will do calibration again, so it works again. Maybe your calibration data was corrupted on some version with NVS bug . @xarrange @shirish47 @kakegu @santanurahaman things you can do to check what's the problem:
1.check the version, give the commit ID here.
2.dump your NVS data to a file, give the file here. 
3.erase the flash and flash again. 
4. check menuconfig about Components config->PHY->Do phy calibration ..., you can do phy calibration every time.

closed 2018-04-04 by FayeY