espressif/esp-idf #975: commit 547ef83 breaks modbus reads

forthlightning, 2017-09-08:
I'm using arduino as component, initializing HardwareSerial( UART_NUM_1 ). Code worked up until this commit.

I communicate with RS485 transceiver linked to UART1. 

I tried to enable the uart with `periph_module_enable(PERIPH_UART1_MODULE);` after Serial1.begin(...), but it uses the wrong baudrate.

If I enable uart1 before hardwareserial, then it hits a wdt reset like:

```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:11360
entry 0x40078cb0
W (283) boot: PRO CPU has been reset by WDT.
W (283) boot: WDT reset info: PRO CPU PC=0x40083a4a
0x40083a4a: _uart_isr at /Users/xx/xx/xx/components/arduino-esp32/cores/esp32/esp32-hal-uart.c:335

W (284) boot: WDT reset info: APP CPU PC=0xae483a1
```

It looks like the arduino library should be setting these bits so that the peripheral timer starts.

forthlightning, 2017-09-08:
EDIT: if i do it this way it works.

```
	DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_UART1_CLK_EN);
	DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_UART1_RST);

	m_m_serial.begin( M_M_BAUDRATE, SERIAL_8N1, M_M_RX, M_M_TX );
```

forthlightning, 2017-09-08:
When I pull latest esp-idf, it hits the WDT as shown above.

igrr, 2017-09-09:
I see that you have also reported this in the arduino-esp32 repository (https://github.com/espressif/arduino-esp32/issues/615). Unless there is something you think should be fixed on IDF side, let's track the issue there.

closed 2017-09-09 by igrr