espressif/esp-idf #979: Missing signature definitions for lwip_ntohs (and similar)

nkolban, 2017-09-09:
There is a macro definition that maps ntohs(x) to lwip_ntohs(x).  Unfortunately, there is no signature definition for lwip_noths().  This results in warnings in the C/C++ editors.  This issue asks for examination and, if warranted, the declaration of the signatures for lwip_ntohs() and related.

zhangyanjiaoesp, 2017-09-22:
@nkolban I'm sorry. Could you show me how to reproduce your problem ?

zhangyanjiaoesp, 2017-09-22:
@nkolban I run an cpp demo which call lwip_ntohs(), but it didn't result in warnings. 
And I found the definition for lwip_ntohs() is in the def.h file, and the definition has already included by "extern "C"". So I don't know why you will result in warning.

closed 2017-10-16 by FayeY