espressif/esp-idf #981: Arduino BLE Client Notification Subscribe Error

WALLTECH, 2017-09-10:
Hello,
In our work and testing over on the ESP Arduino repo, I've run into a debug line that's causing any reading of characteristics or their notifications to be ignored and I don't want them to. How can this be fixed? I'm trying to read from a known working peripheral when used with curieBLE or nRF connect.

E (51039) BT: bta_gattc_cmpl_cback unknown conn_id = 3, ignore data

Weijian-Espressif, 2017-09-26:
hi @WALLTECH , please give me more log information to locate the problem.

nkolban, 2017-09-27:
Over the last week, the ESP-IDF BLE API changed and that required rework of our Arduino BLE libraries.  We have done that but are now suffering an inability to compile C++ applications because of an apparent problem in the latest toolchain (see #1032).  Unfortunately until we can compile C++ classes as described in that issue, we are going to be somewhat stuck in giving you more log diagnostics (I think).

FayeY, 2017-11-21:
Any update about this issue?

closed 2017-12-01 by FayeY