espressif/esp-idf #983: esp_deep_sleep triggers WDT core reset

forthlightning, 2017-09-10:
I put the following in app main

```
    struct timeval now;
    gettimeofday(&now, NULL);
    
    const int wakeup_time_sec = 2;
    ESP_LOGI( __func__, "Entering deep sleep\n" );
    esp_deep_sleep(wakeup_time_sec * 1000000LL);
    ESP_LOGI( __func__, "Exit deep sleep\n" );
```

And get the following on serial

```

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4384
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:11416
entry 0x40078cd0
I (211) cpu_start: Pro cpu up.
I (211) cpu_start: Single core mode
I (213) heap_init: Initializing. RAM available for dynamic allocation:
I (223) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (241) heap_init: At 3FFC2C08 len 0001D3F8 (116 KiB): DRAM
I (260) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (280) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (300) heap_init: At 40087B3C len 000184C4 (97 KiB): IRAM
I (319) cpu_start: Pro cpu start user code
I (388) cpu_start: Starting scheduler on PRO CPU.
I (394) app_main: Entering deep sleep

ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4384
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:11416
entry 0x40078cd0
W (48) boot: PRO CPU has been reset by WDT.
W (48) boot: WDT reset info: PRO CPU PC=0x63dae3c0
W (49) boot: WDT reset info: APP CPU PC=0x6dcbbb15
I (229) cpu_start: Pro cpu up.
I (229) cpu_start: Single core mode
I (231) heap_init: Initializing. RAM available for dynamic allocation:
I (241) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (260) heap_init: At 3FFC2C08 len 0001D3F8 (116 KiB): DRAM
I (279) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (298) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (318) heap_init: At 40087B3C len 000184C4 (97 KiB): IRAM
I (337) cpu_start: Pro cpu start user code
I (406) cpu_start: Starting scheduler on PRO CPU.
I (412) app_main: Entering deep sleep

ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
```

Over and over again. :(

Am I missing something?

negativekelvin, 2017-09-10:
https://github.com/espressif/esp-idf/issues/964

igrr, 2017-09-11:
@negativekelvin No, probably not the same. That one is WDT reset instead of deep sleep reset. This one is WDT reset *after* deep sleep reset.

Please try increasing the delay after deep sleep wakeup in menuconfig, see http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#esp32-deep-sleep-wakeup-delay.

forthlightning, 2017-09-11:
That did it! Thanks. I just arbitrarily picked the midrange value, (2500) any reason to choose something else? If not feel free to close.

Spritetm, 2017-09-12:
Essentially, the thing is that your flash chip doesn't wake up quickly enough to allow the ESP32 to read from it; the value is the time the ESP32 will wait (2.5ms in this case). Increasing it makes the ESP32 very slightly slower in waking up, but should otherwise be harmless. If 2500 works for you, then go for it :)

closed 2017-09-29 by forthlightning