espressif/esp-idf #985: how to configurate the menuconfig ? if I want to use larger memory for init data in program.

snowQY, 2017-09-11:
I want to send a picture to my compute  by esp32，but the picture is 40KB，i need to define a 66222 bytes
array in program to save the picture  which from the ARM .compile has not error,but the esp32 will restart while the client   connect to the esp32 service ,and  after relink ,message:
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr 
Register dump:
PC      : 0x4010c4b7  PS      : 0x00060036  A0      : 0x8010ceb4  A1      : 0x3ffb9000  
A2      : 0x0000000a  A3      : 0xffffffe0  A4      : 0x00000000  A5      : 0x3ffa8cb0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x800fffd2  A13     : 0x3ffa8c80  
A14     : 0x00000000  A15     : 0x3ffa8cc0  SAR     : 0x0000000b  EXCCAUSE: 0x00000001  
EXCVADDR: 0xffffffe0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
.But it is ok ,if i use 2kb picture and smaller array.So i guess there are not enough memory to allocte for the array.

Spritetm, 2017-09-11:
You do not specify the log lines before this error... I think you ran into a stack issue. How do you allocate the 66K array? Statically, or using malloc()? If the first, suggest increasing the amount of stack allocated to your task, or even better: modifying your code to use malloc() instead.

snowQY, 2017-09-11:
Thank you very much for your advices.I allocate the array statically.one minute ago,i use malloc() to try,but it also has a error , hint some strange characters ,and you suggested me to increas the amount of stack ,but i don't know how to do,can you tell me more about it?

Spritetm, 2017-09-12:
"but it also has a error , hint some strange characters" - erm, that's not really helpful. What error do you get, what strange characters?

It would be beneficial if you could also post the code you use; you may be doing something wrong without realizing this and we may be debugging entirely the wrong thing.

snowQY, 2017-09-12:
Thank you a lot,i set a 921600 baud rate,but the serial tool support max is 230400,so some strange characters appear,now it is ok.My goal is sending video to my compute with esp32,and our board is inserted esp32,but only uart0 is connected to FPGA GPIO ，and the PCB is can not be changed,because of some reasons,so the video datas only be sended to wifi TX buffer via uart0,but the max baud rate 921600 is so low for the video.So can you have some better advices? or some demo for video transmission or ping-pong uart buffer.Thank you again for your time and attention,and i sincerely looking forward to hear from you soon.

igrr, 2017-12-19:
ESP32 UART supports higher baud rates than 921600, you can use UART driver to set the baud rate you need. Please check http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/uart.html.

igrr, 2018-01-08:
Please feel free to reopen if there is still an issue.

closed 2018-01-08 by igrr