espressif/esp-idf #986: Faster Wifi station connect improvement - avoid 100ms passive scan

gatispei, 2017-09-11:
Hello,

As recommended by ESP_Sprite on esp32.com forum I am opening an issue on this here:

How would it be possible to setup wifi station mode so that ESP32 connects to AP immediately after receiving good Probe Response?

My tests are showing that after esp_wifi_connect() is invoked (sta.channel in config is set to channel the AP is in) the ESP32 actually does so called "active scan" by sending Probe Request frame and receives a good "Probe Response".

But the problem is that ESP32 even after receiving good Probe Response is still waiting for the per channel scan timeout (100ms) before attempting to actually connect.

I can understand the logic behind this behaviour of course - there might be other APs with same SSID, with better signal that might respond later.

But it would be greatly appreciated if it would be possible to have a "hinted" connect mode - just like channel "hint" in struct wifi_sta_config_t - connect to first responding AP (or connect to AP with specific MAC address avoiding scanning / Probe Request / Response stuff altogether).

This would save about 100ms of PHY RF active time in most cases when the AP you are connecting to is the same as previous time -
which is a great deal of power saving in our battery powered application.

Gatis

gatispei, 2017-09-17:
Just to illustrate the issue I am attaching an oscilloscope screenshot. The graph shows current drawn over time in a typical esp32 boot in our application.

On x axis is time - 100ms per division, on y axis is amperes 100mA per division (actually volts but I am using a special device that converts mA to mV.

As you can see the passive scan time contributes a big part of the overall current consumption of the device in a typical boot.

[image]

Please advise on how it would be possible to implement the above feature. Thank you.

liuzfesp, 2017-09-18:
Thanks @gatispei, we will put it into our optimization list. Now some optimizations are already done (but merge into IDF master yet), and more optimized options for scan will be supported in release 3.0.

gatispei, 2017-09-18:
Great, sounds good! Are there any time estimates for v3.0 release?

kimata, 2017-11-22:
Is there any progress?

jack0c, 2018-01-14:
@gatispei @kimata We are still considering about this. Will add a flag(temporary names 'Stop on first match') in scan. But we have no time to merge this into idf3.0, which was feature freezed. It will be added into idf3.1. The good news is, after scanning refactoring in idf3.0, it will be easy to do this.

gatispei, 2018-01-16:
@jack0c @liuzfesp We are still interested in the feature, thanks for the update!

Stasikiii, 2018-10-09:
@jack0c still no any news?
esp8266 can connect to AP for 200ms but esp32 for 900 :(

Abishek05, 2019-09-29:
@jack0c Any updates on this improvement?

espxiehang, 2019-11-15:
We recorded this issue and plan to do it later.

espxiehang, 2020-11-10:
I'm very sorry,After our internal discussion,we decided not to support this feature.

KaeLL, 2020-11-10:
> I'm very sorry,After our internal discussion,we decided not to support this feature.

Could you elaborate on why?

espxiehang, 2020-11-30:
Hi @KaeLL ,We can support this feature.
I will inform you when this function is completed.

closed 2021-01-05 by espressif-bot