espressif/esp-idf #988: pthreads and setting stack size

nkolban, 2017-09-12:
I fully realize that the pthreads support is still being baked ... this issue may be in plan already for 3.0.

I would like to see the stack size API for pthreads implemented.  Specifically, I'd like to see the stack size attribute implemented.  Specifically those APIs called pthread_attr_setstacksize() and pthread_attr_getstacksize().  I commonly find that I want to tune my stack sizes for FreeRTOS tasks and the ability to set the stack size before the pthread is created is desired.  Looking at the code, it doesn't appear like it would be horrible to implement (guess).

See:
http://man7.org/linux/man-pages/man3/pthread_attr_setstacksize.3.html

PerMalmberg, 2017-10-19:
+1 for this.

snej, 2018-06-28:
As a workaround, you can use the API calls in [esp_pthread.h](https://github.com/espressif/esp-idf/blob/master/components/pthread/include/esp_pthread.h) to set the default stack size for pthreads.

igrr, 2018-11-19:
`pthread_attr_setstacksize` has been implemented in https://github.com/espressif/esp-idf/commit/f27db1f24194f9a40f4a9d50f3471160c645f4fb.

closed 2018-11-19 by igrr