espressif/esp-idf #989: Support for WiFi 802.11 Ad Hoc mode

nickzoic, 2017-09-12:
Is it possible to add support for WiFi 802.11 Ad Hoc mode to the ESP32 WiFi Stack Libraries?

honza-klu, 2017-09-12:
+ One for me. Ad Hoc mode would be extremely useful for p2p comunication and mesh networks.

jack0c, 2018-01-03:
Sorry, @nickzoic Ad Hoc will not be in our TODO list. @honza-klu P2P(Wi-Fi DIRECT) is  in our TODO list. And , we will release ESP-Mesh in idf3.1.

nickzoic, 2018-01-03:
No worries.  I've been experimenting a bit with ESP-Now and that's looking good for my application.
I'm looking forward to trying ESP-Mesh!

Closing this ...

closed 2018-01-03 by nickzoic

wgaylord, 2019-11-07:
Will Ad-Hoc ever be on the ToDo list? Really want to use my ESPs with existing mesh networks that follow the 802.11 ad-hoc standard for making mesh networks.

AndroidBaaaby, 2019-11-08:
Hi @wgaylord . Ad Hoc is not in our TODO list. Sorry for that.

wgaylord, 2019-11-08:
Is there a specific reason why? Or just other priorities?

preechaim, 2019-12-06:
As far as I know, the Ad-Hoc (IBSS) mode is basic functionality in the WiFi stack. 
It should not be difficult to expose this function to the framework.
So please reconsider.

AndroidBaaaby, 2019-12-17:
Hi @honza-klu @preechaim, thank you for your suggestion. For now we are doing other tasks. If there are any news about this, we would let you know.

wgaylord, 2020-08-07:
Is 802,11 Ad Hoc going to be looked at or is it not being looked at as it would compete with ESP-Mesh? Just looking for an answer.  I would attempt to implement Ad Hoc somehow but the WiFi stack is closed source in a binary only so I would have to use sending and reception of raw packets.