espressif/esp-idf #991: Build with -Wundef

rojer, 2017-09-12:
Currently ESP-IDF is "`-Wundef` dirty", that is, macros are evaluated for value without having been previously defined.
This prevents us from enabling this useful check on our own code. Even if not enabling by default, ESP-IDF should be made `-Wundef`-clean

rojer, 2017-09-12:
example errors:

```
/opt/Espressif/esp-idf/components/newlib/include/sys/reent.h:458:5: error: "DEBUG" is not defined [-Werror=undef]
 #if DEBUG
     ^

/opt/Espressif/esp-idf/components/newlib/include/stdio.h:239:5: error: "__GNU_VISIBLE" is not defined [-Werror=undef]
 #if __GNU_VISIBLE
     ^

/opt/Espressif/esp-idf/components/esp32/include/xtensa/config/core.h:809:38: error: "XCHAL_ERRATUM_453" is not defined [-Werror=undef]
 #if (XCHAL_LOOP_BUFFER_SIZE == 0) || XCHAL_ERRATUM_453

/opt/Espressif/esp-idf/components/freertos/include/freertos/FreeRTOSConfig.h:254:41: error: "CONFIG_SUPPORT_STATIC_ALLOCATION" is not defined [-Werror=undef]
 #define configSUPPORT_STATIC_ALLOCATION CONFIG_SUPPORT_STATIC_ALLOCATION

```

etc.

lws-team, 2017-09-12:
Oh... that is is pretty cool... I never heard of it until now.

projectgus, 2017-09-13:
Hi @rojer,

This feels like the --warn-undef Makefile change all over again. ;)

One thing to note, none of the example errors you posted are mistakes. The "fix" for all of them is probably to change something like ie `#if DEBUG` to `#if defined(DEBUG) && DEBUG`, which seems unnecessarily verbose for the same end result.

You are right that maybe there are mistakes that this will catch though (did you happen to find any in the IDF codebase?)

Also, this preprocessor behaviour (undefined macro expands to nothing) gets relied on in a bunch of IDF's third party code - where we can't easily patch it.

For now, suggest adding:
```
CPPFLAGS += -Wundef
```
To component.mk for any components that you have which are `-Wundef`-clean.

projectgus, 2017-09-13:
> For now, suggest adding:
> ```
> CPPFLAGS += -Wundef
> ```

As I posted it I realised this isn't a great suggestion because lots of the warnings come from public headers. Hmm.

lws-team, 2017-09-13:
At least some of the spew I get looks like it should be fixed, eg

```
In file included from /projects/esp-idf/components/esp32/include/xtensa/xtruntime-frames.h:30:0,
                 from /projects/esp-idf/components/freertos/include/freertos/xtensa_context.h:47,
                 from /projects/esp-idf/components/freertos/include/freertos/xtensa_api.h:33,
                 from /projects/esp-idf/components/esp32/include/esp_intr.h:19,
                 from /projects/esp-idf/components/driver/include/driver/touch_pad.h:20,
                 from /projects/esp-idf/components/esp32/include/esp_sleep.h:20,
                 from /projects/esp-idf/components/esp32/include/esp_system.h:21,
                 from /projects/esp-idf/components/lwip/include/lwip/port/lwipopts.h:40,
                 from /projects/esp-idf/components/lwip/include/lwip/lwip/opt.h:45,
                 from /projects/esp-idf/components/lwip/include/lwip/lwip/netdb.h:32,
                 from /projects/esp-idf/components/lwip/include/lwip/posix/netdb.h:33,
                 from /projects/esp32/lws-esp32-factory/components/libwebsockets/lib/private-libwebsockets.h:172,
                 from /projects/esp32/lws-esp32-factory/components/libwebsockets/lib/base64-decode.c:43:
/projects/esp-idf/components/esp32/include/xtensa/config/core.h:809:38: error: "XCHAL_ERRATUM_453" is not defined [-Werror=undef]
 #if (XCHAL_LOOP_BUFFER_SIZE == 0) || XCHAL_ERRATUM_453
```

it should have defined() checks if there's a way for them not to be defined.

rojer, 2017-09-13:
> This feels like the --warn-undef Makefile change all over again. ;)

i know, i know :)

see, this came out of real bug - admittedly, in my codebase, but i basically fat-fingered macro name in an `#if` and spent some quality time debugging, since the code block it controlled, and which was not enabled as a result of the typo, broke things in non-obvious ways. i was very surprised to find that `-Wundef` is not part of `-Wall`, but turns out it's not, and we explicitly enable `-Wundef` on our other arches.

i think `-Wundef` is a good default, but if not, IDF should at least be made clean. c'mon guys, ESP8266 RTOS _and_ non-OS SDKs are undef-clean, i know IDF can do better :)

rojer, 2017-09-13:
> The "fix" for all of them is probably to change something like ie #if DEBUG to #if defined(DEBUG) && DEBUG, which seems unnecessarily verbose for the same end result.

either that, or introduce a common header that `#ifndef`/`#define`s all the tunables, e.g. (let's call it `tunables.h` or whatever):
```
#ifndef DEBUG
#define DEBUG 0
#endif
... etc ...
```

 it will also serve as a manifest of things that can be tweaked, which would be a good thing.

projectgus, 2017-09-13:
> i was very surprised to find that -Wundef is not part of -Wall

I think this is because so much legacy code relies on the undefined-expands-empty behaviour of the C preprocessor. There are a lot of nice gcc features like this *looks wistfully into the distance*.

But point taken. :)

>it should have defined() checks if there's a way for them not to be defined.

Sure, but this doesn't indicate an actual bug. Xtensa only define `XCHAL_ERRATUM_453` if that errata value is set, and `#if (1 == 0) ||` is a valid preprocessor expression (albiet an ugly one.)

> it should have defined() checks if there's a way for them not to be defined.

Perhaps, but we have to be sure not to cause a regression. If we use `defined(XCHAL_ERRATUM_453)` somewhere and some future upstream Xtensa header release comes with `#define XCHAL_ERRATUM_453 0` then the old "bad" code will continue to work correctly but the new "good" code will quietly start doing the wrong thing.

lws-team, 2017-09-13:
> f we use defined(XCHAL_ERRATUM_453) somewhere and some future upstream Xtensa header

No I mean

```
#if (XCHAL_LOOP_BUFFER_SIZE == 0) || (defined(XCHAL_ERRATUM_453) && XCHAL_ERRATUM_453)
```

It should be possible to talk to the Xtensa dudes about this and try to get them to take the fixes.

projectgus, 2017-09-13:
> No I mean

Please read what I wrote about this pattern a few replies before you pointed out XCHAL_ERRATUM_453:

> ie #if DEBUG to #if defined(DEBUG) && DEBUG, which seems unnecessarily verbose for the same end result.

I'm not saying we won't do this, I'm just saying that we have to think about the implications carefully - especially when it comes to editing a lot of third party code.

jcmvbkbc, 2017-09-13:
> and `#if (1 == 0) ||` is a valid preprocessor expression (albeit an ugly one.)

It is not. If you try to compile it you get the following: `error: operator '||' has no right operand`. In `#if`/`#elif` preprocessor expressions undefined identifiers are replaced with 0 according to C standards (e.g. for C99 see 6.10.1:3).

lws-team, 2017-09-13:
You just say it's "unnecessarily verbose" in your opinion.  But since the extensa core includes don't really control how they are included they should be defensive and explicit... IMO...

> I'm not saying we won't do this, I'm just saying that we have to think about the implications carefully - especially when it comes to editing a lot of third party code.

OK, but there seems some stuff closer to home.

```
/projects/esp-idf/components/esp32/include/esp_wifi.h:132:5: error: "CONFIG_ESP32_WIFI_NVS_ENABLED" is not defined [-Werror=undef]
 #if CONFIG_ESP32_WIFI_NVS_ENABLED
     ^
/projects/esp-idf/components/esp32/include/esp_wifi.h:138:5: error: "CONFIG_NEWLIB_NANO_FORMAT" is not defined [-Werror=undef]
 #if CONFIG_NEWLIB_NANO_FORMAT
     ^
/projects/esp-idf/components/spi_flash/include/esp_spi_flash.h:336:5: error: "CONFIG_SPI_FLASH_ENABLE_COUNTERS" is not defined [-Werror=undef]
 #if CONFIG_SPI_FLASH_ENABLE_COUNTERS
```

Do you want patches for these kind of problems in esp-idf balliwick?

rojer, 2017-09-13:
`CONFIG_*` vars are generated by Kconfig and put into `sdkconfig.h`. currently entries are not emitted for settings that are not enabled, but they can be, with a simple change.

rojer, 2017-09-13:
currently IDF headers prevent me from turning on `-Wundef` on my code. i think that should definitely be fixed, and that's all i really care about. but i do find it immensely helpful during development, i think you will find it useful too.

projectgus, 2017-09-13:
@rojer Fair enough. I can see it will help with catching typos and the like in `#if` expressions. We just have to think about whether it's worth the additional work to make the change and maintain it (mostly I'm thinking of other people's code, unlike the Makefile `--warn-undef` change where it was all internal to IDF).

lws-team, 2017-09-13:
-Wundef looks to me to be valuable... it found several problems on lws when I enabled it.  Eg making decisions for `OPENSSL_VERSION_NUMBER` when the build was configured for MBEDTLS which doesn't define it.  The silence of defaulting to interpreting undefined as a value hides these without -Wundef.

However if you don't think the problems are problems, consequently you won't see any value in it :-)

projectgus, 2017-09-13:
Thank you for explaining a place where this would have exposed a problem for you, @lws-team . I'm not suggesting there are no problems related to undefined macros, just that most (all?) of the warnings that appear in IDF when you turn this option on are not "real" problems (as in, the behaviour was correct before it was turned into a warning and any change will be to satisfy the warning not to fix incorrect behaviour in the program.)

m-bruck, 2020-05-23:
Your discussion of the XCHAL_ERRATUM_453 warning overlooks the fact that this macro is _always_ defined (same file line 1379). Unfortunately that is _after it is used_ (line 809).

rojer, 2025-07-14:
@igrr that's great news, Ivan! does that mean patches to that effect will be accepted?

igrr, 2025-07-14:
@rojer Of course, patches are appreciated. However I think we need to first figure out how to check that IDF public headers are Wundef-clean, only then will the fixes to the actual headers be meaningful.

Enabling -Wundef in IDF's own CI builds (as we currently do for -Werror, https://github.com/espressif/esp-idf/blob/bfe5caf58f742fd35c023335f475114a5b88761e/tools/ci/configure_ci_environment.sh#L20) wouldn't work as it would require not only the headers but all the source files to be Wundef-clean.

Enabling Wundef inside our public header checker script (https://github.com/espressif/esp-idf/blob/bfe5caf58f742fd35c023335f475114a5b88761e/tools/ci/check_public_headers.py#L188) would only catch the issue if the Kconfig option being used in an `#if CONFIG_XYZ` condition is undefined by default. Otherwise, we would have to run the public header checker for many possible sets of Kconfig options.

Simply grepping for `#if CONFIG_` in the public header files might do the trick, but not sure if this approach is robust enough?

igrr, 2025-07-14:
That would work, but that has the problem that errors will only be reported if the CONFIG_XYZ option used in the `#if CONFIG_XYZ` condition is not set. If this option is set, no error would be reported. This means that in order for this check to be effective, this app needs to have a carefully crafted sdkconfig.defaults which disables every option which is ever used in public header files. This seems very error prone.

(This is the same problem as the one with the public header check script mentioned above.)