espressif/esp-idf #992: PDM stop working after some amount of time.

fanthos, 2017-09-12:
I am trying to build a Bluetooth audio receiver with https://github.com/MrBuddyCasino/ESP32_MP3_Decoder , but I found that if I enabled Bluetooth and PDM at the same time, the music stop playing after several seconds. It always stop working at the same time after I press Play button.

Also tested WiFi MP3 player with PDM, it stop playing after about 40 minutes.
Bluetooth speaker works great if I disable PDM and use internal DAC instead.

I got logs like this:
```
E (14625) BT: BTA_AV_START_EVT status 0, suspending 0, init 0
I (14635) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x1
I (14635) BT_AV: a2dp audio_state_cb state 2
I (16795) BT_AV: RAM left: 121092
E (16795) BT_AV: audio data pkt cnt 100
I (18975) BT_AV: RAM left: 114172
E (18975) BT_AV: audio data pkt cnt 200
I (21155) BT_AV: RAM left: 106804
E (21155) BT_AV: audio data pkt cnt 300
I (23335) BT_AV: RAM left: 98008
E (23335) BT_AV: audio data pkt cnt 400
I (25515) BT_AV: RAM left: 90316
E (25515) BT_AV: audio data pkt cnt 500
I (27695) BT_AV: RAM left: 81732
E (27695) BT_AV: audio data pkt cnt 600
I (29885) BT_AV: RAM left: 118484
E (29885) BT_AV: audio data pkt cnt 700
```
I dont have a I2S dac, but if I change the code to use DAC it will keep playing.
If I comment out https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/blob/master/components/audio_renderer/audio_renderer.c#L122-L126 , Bluetooth works great but no sound.

After the bluetooth stop working, my phone shows disconnected from the board several seconds later.
```
E (46015) BT: bta_av_str_stopped:audio_open_cnt=1, p_data 0
E (46015) BT: bta_dm_rm_cback:0, status:6
E (46015) BT: L2CAP - no LCB for L2CA_SetAclPriority
E (46025) BT: WARNING L2CA_SetFlushTimeout No lcb for bd_addr [...;32f987]
E (46025) BT: bta_dm_rm_cback:0, status:1
E (46035) BT: ccb 0 not allocated
```

I added memory monitoring task on my fork(no other modifications)
https://github.com/fanthos/ESP32_MP3_Decoder

FayeY, 2018-03-20:
Hi, so sorry that we cannot reproduce your problem, "i2s stop work" does it mean that there is no signal output from the I2S?

fanthos, 2018-03-20:
This happened when using PDM mode of I2S, using it as Bluetooth speaker makes it faster occurred.
The issue is like sound glitch, repeating same sound(about less than 1 second). With newer SDK, the log shows one task does not return.

fanthos, 2018-03-20:
I just upload the newer code. Change `#define AUDIO_OUTPUT_MODE 2` to 3 will reproduce the issue.