espressif/esp-idf #993: make flash problem

JPMJPM, 2017-09-12:
Hi,

I am trying to install esp-idf in Windows 10 but "make flash" has this problem :

mainuser@mainpc MINGW32 ~/esp/hello_world
$ make flash
Traceback (most recent call last):
  File "C:/msys32/home/mainuser/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 21, in <module>
    import argparse
  File "C:\Python27\Lib/argparse.py", line 85, in <module>
    import collections as _collections
  File "C:\Python27\Lib/collections.py", line 20, in <module>
    from _collections import deque, defaultdict
ImportError: No module named _collections
make[1]: *** [/home/mainuser/esp/esp-idf/components/esptool_py/Makefile.projbuild:51: /home/mainuser/esp/hello_world/build/bootloader/bootloader.bin] Error 1
make: *** [/home/mainuser/esp/esp-idf/components/bootloader/Makefile.projbuild:40: /home/mainuser/esp/hello_world/build/bootloader/bootloader.bin] Error 2

Any help ?.

Thanks in advance and regards.

projectgus, 2017-09-13:
Hi @JPMJPM ,

It looks like IDF is pulling in a different Python install, an external one at C:\Python27. Then this is failing to import a Python built-in module, I don't know why that would be - maybe that external Python install is broken somehow?

You run "make menuconfig" and try changing "Python 2 Interpreter" under "SDK Tools Configuration" to "/mingw32/bin/python", does it then work?

JPMJPM, 2017-09-13:
Thanks projectgus,

With "/mingw32/bin/python" in "SDK Tools Configuration" it does not work but with
"C:/Python27/python" everything work fine.

Regards.

projectgus, 2017-09-13:
Glad you found a solution. I'm guessing this has something to do with the PATH environment variable. To help us see if we can work around this for other users, could you please run the command ` env | grep -i python` in the MINGW32 terminal and paste the output here?

JPMJPM, 2017-09-13:
mainuser@mainpc MINGW32 ~/esp
$ env | grep -i python
PYTHONHOME=C:\Python27
PYTHONPATH=C:\Python27\Lib

mainuser@mainpc MINGW32 ~/esp
$

fanthos, 2017-09-14:
How did start the shell? Run mingw32.exe or run the bat with -mingw32 argument should work.

JPMJPM, 2017-09-14:
I start the shell running a desktop shortcut with target argument "C:\msys32\mingw32.exe".

fanthos, 2017-09-14:
```
echo $PATH
env | grep MSYS
env | grep MINGW
```
`/mingw32/bin/python` works great on my device.

projectgus, 2017-09-14:
@fanthos It's the external Python install and (I think) PYTHONPATH environment variable which makes the difference here.

@JPMJPM If you want to fix this problem without needing to change sdkconfig for each project then you can profile add a line "unexport PYTHONPATH" to /etc/profile or ~/.profile inside the MSYS2 environment. Then it will use the bundled Python for everything and not call out to C:\Python27.

However, if you're happy with the solution you already have then that should be fine too.

JPMJPM, 2017-09-14:
Hi projectgus,

* unexport PYTHONPATH in /etc/profile ---> unexport, command not found
* unexport PYTHONPATH in ~/.profile ---> the known error ImportError: No module named _collections

projectgus, 2017-09-26:
Thanks @JPMJPM for the update. Sorry for the slow reply.

If you open the shell and run:
```
echo $PATH
echo $PYTHONPATH
```
... what is printed?

projectgus, 2017-09-26:
Thanks @JPMJPM  for the update. Sorry for the slow reply.

To be honest, I don't fully understand how Windows environment varialbes are exported into MSYS2. Maybe something like:
```
export PYTHONPATH=""
```
will work better?

fanthos, 2017-09-26:
I think the command should be `unset PYTHONPATH` or something.

closed 2017-11-21 by FayeY