espressif/esp-idf #994: SD-SPI file writes seem to corrupt the heap

tetious, 2017-09-13:
Executing a simplified version of the storage sample seems to immediately cause heap corruption in current master. I'm not 100% sure what commit is to blame, but it was fairly recent. Possibly even the SPIFFS commit. I can try to narrow it down further if needed.

Here is the code I'm running, and the crash happens on file.close()
```c++
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    host.max_freq_khz = SDMMC_FREQ_PROBING;
    sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_DEFAULT();
    slot_config.gpio_miso = static_cast<gpio_num_t>(spi_miso);
    slot_config.gpio_mosi = static_cast<gpio_num_t>(spi_mosi);
    slot_config.gpio_sck = static_cast<gpio_num_t>(spi_clk);
    slot_config.gpio_cs = static_cast<gpio_num_t>(sd_spi_ss);

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
            .format_if_mount_failed = false,
            .max_files = 5
    };

    esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sd", &host, &slot_config, &mount_config, nullptr);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            log_e("Failed to mount filesystem.");
        } else {
            log_e("Failed to initialize the card (%d). ", ret);
        }
    }

    ofstream file;    
    file.open("/sd/test.file.omg", ios::out);
    if(!file)
        log_e("Open failed.");
    else
        log_i("Opened");
    file << "zombie";
    log_i("written");
    file.close(); 
    log_i("closed");

``` 

The crash:
```
CORRUPT HEAP: Bad tail at 0x3ffc91c1. Expected 0xbaad5678 got 0xba000000
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c", line 195, function: multi_heap_free
abort() was called at PC 0x40106f6b on core 1

Backtrace: 0x40087524:0x3ffcaec0 0x40087623:0x3ffcaee0 0x40106f6b:0x3ffcaf00 0x40087265:0x3ffcaf30 0x4008302e:0x3ffcaf50 0x4008347d:0x3ffcaf70 0x4000bec7:0x3ffcaf90 0x400fbbe2:0x3ffcafb0 0x400fbc0e:0x3ffcaff0 0x400fa393:0x3ffcb020 0x400fa757:0x3ffcb050 0x400fac71:0x3ffcb090 0x400fae1f:0x3ffcb0d0 0x40110015:0x3ffcb110 0x40104bc3:0x3ffcb130 0x40105701:0x3ffcb190 0x400ff803:0x3ffcb1c0 0x400ff6be:0x3ffcb1e0 0x400ffa76:0x3ffcb200 0x400ffac8:0x3ffcb220 0x40100096:0x3ffcb240 0x401001cc:0x3ffcb260 0x40101239:0x3ffcb280 0x4010278b:0x3ffcb2c0 0x40105d22:0x3ffcb2f0 0x4000bd83:0x3ffcb310 0x4000182a:0x3ffcb330 0x400de75d:0x3ffcb350 0x400de847:0x3ffcb370 0x400d338d:0x3ffcb390 0x400d37dc:0x3ffcb3e0 0x400d3489:0x3ffcb420 0x400d353b:0x3ffcb4e0 0x400d36fd:0x3ffcb500 0x400d1968:0x3ffcb520 0x4010629f:0x3ffcb690
```

Decoded:
```
0x0000000c: ?? ??:0
0x0000000a: ?? ??:0
0x00000000: ?? ??:0
0x000000ab: ?? ??:0
0x00000000: ?? ??:0
0x00000bac: ?? ??:0
0x4008302e: heap_caps_free at heap_caps.c:172
0x3ffcaff0: ?? ??:0
0x40110015: sdmmc_send_cmd at sdmmc_cmd.c:350
0x3ffcb1e0: ?? ??:0
0x40101239: f_write at ff.c:4443
0x3ffcb330: ?? ??:0
0x400d3489: std::basic_filebuf<char, std::char_traits<char> >::_M_terminate_output() at fstream:758
0x3ffcb690: ?? ??:0
```

igrr, 2017-09-13:
Can you please upload your elf file somewhere? The "decoded" backtrace seems to be slightly odd (first six lines are clearly invalid), but the original backtrace is okay.

tetious, 2017-09-13:
Sure. I wish I knew what I was doing wrong with xtensa-esp32-elf-addr2line. :(

[firmware.elf.zip](https://github.com/espressif/esp-idf/files/1298081/firmware.elf.zip)

igrr, 2017-09-13:
Ok, so here's the backtrace:
```
0x40087524: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c:553
0x40087623: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c:553
0x40106f6b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x40087265: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c:195 (discriminator 1)
0x4008302e: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c:172
0x4008347d: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c:33
0x4000bec7: ?? ??:0  (this is "free" in ROM)
0x400fbbe2: spi_device_get_trans_result at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./spi_master.c:685
0x400fbc0e: spi_device_transmit at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./spi_master.c:699
0x400fa393: poll_response_token at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./sdspi_host.c:470
0x400fa757: start_command_write_blocks at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./sdspi_host.c:772
0x400fac71: sdspi_host_start_command at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./sdspi_host.c:382
0x400fae1f: sdspi_host_do_transaction at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./sdspi_transaction.c:86
0x40110015: sdmmc_send_cmd at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/sdmmc/./sdmmc_cmd.c:350
0x40104bc3: sdmmc_write_sectors_dma at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/sdmmc/./sdmmc_cmd.c:342 (discriminator 4)
0x40105701: sdmmc_write_sectors at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/sdmmc/./sdmmc_cmd.c:725
0x400ff803: ff_sdmmc_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/diskio_sdmmc.c:51
0x400ff6be: ff_disk_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/diskio.c:73
0x400ffa76: sync_window at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/ff.c:4443
0x400ffac8: move_window at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/ff.c:4443
0x40100096: get_fat at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/ff.c:4443
0x401001cc: create_chain at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/ff.c:4443
0x40101239: f_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/ff.c:4443
0x4010278b: vfs_fat_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/vfs_fat.c:298
0x40105d22: esp_vfs_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/vfs/./vfs.c:242 (discriminator 3)
0x4000bd83: ?? ??:0
0x4000182a: ?? ??:0
0x400de75d: _ZN12_GLOBAL__N_1L6xwriteEiPKci at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/src/c++98/basic_file.cc:211
0x400de847: _ZNSt12__basic_fileIcE6xsputnEPKci at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/src/c++98/basic_file.cc:211
0x400d338d: _ZNSt13basic_filebufIcSt11char_traitsIcEE22_M_convert_to_externalEPci at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/fstream:758
0x400d37dc: _ZNSt13basic_filebufIcSt11char_traitsIcEE8overflowEi at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/fstream:758
0x400d3489: _ZNSt13basic_filebufIcSt11char_traitsIcEE19_M_terminate_outputEv at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/fstream:758
0x400d353b: _ZNSt13basic_filebufIcSt11char_traitsIcEE5closeEv at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/fstream:758
0x400d36fd: _ZNSt14basic_ofstreamIcSt11char_traitsIcEE5closeEv at /home/ivan/e/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/fstream:758
0x400d1968: _Z5setupv at /home/greg/Code/esp32-test-3/src/main.cpp:51
0x4010629f: _Z8loopTaskPv at /home/greg/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/main.cpp:15
```

ginkgm, 2017-09-13:
hi @tetious, I think we've catched where this happens:

1. [The SPI driver](https://github.com/espressif/esp-idf/blob/63e1e4e502178ced9d29999e45050c952be29c26/components/driver/spi_master.c#L624) incorrectly allocated a buffer that is not 32-bit aligned for DMA. The buffer for DMA should always start from 32-bit boundary and have length of multiples of 4 bytes.
2. [This piece of code](https://github.com/espressif/esp-idf/blob/843159b0d7284ba560345af312a1e7f8b14d19e7/components/driver/sdspi_host.c#L456) made the spi driver allocate buffer incorrectly. This is not the root cause, but you can fix this by using SPI_TRANS_USE_RXDATA to improve its effciency. 

We're currently working on this and doing tests.

closed 2017-09-21 by igrr

ginkgm, 2017-09-22:
@tetious we updated the spi and sdspi driver to solve this issue, please have a try.

If the issue still occurs, please come back for help.

commit 074f36af (2017-09-21): fix(spi_master): fix the heap corruption bug that RX DMA writes over the temporary buffer boundary.

TW#15434

Closes #994.