espressif/esp-idf #995: iPhone 7 as BLE Central causes reset of ESP32 device

SiarheiVolkau, 2017-09-13:
We are encounter this problem every time when iPhone 7 / 7+ connects to ESP32:

`Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:`
`PC      : 0x40047de7  PS      : 0x00060b30  A0      : 0x80049b9c  A1      : 0x3ffb70f0  `
`A2      : 0x07ffffff  A3      : 0x00000013  A4      : 0x00002bf0  A5      : 0x00000001  `
`A6      : 0x3ffb7150  A7      : 0x00000000  A8      : 0x3ffbb710  A9      : 0x3ffb70d0  `
`A10     : 0x0030bb00  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffafd68  `
`A14     : 0x00000000  A15     : 0x3ffb8360  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  `
`EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  `

`Backtrace: 0x40047de7:0x3ffb70f0 0x40049b99:0x3ffb7150 0x400457cd:0x3ffb7180 0x4001a637:0x3ffb71c0 0x40019d11:0x3ffb71f0 0x40055b4d:0x3ffb7210 0x401729ef:0x3ffb7230 0x40172a3f:0x3ffb7250`

error happened in closed source library, so we can't understand what cause this problem.

ESP32 is BLE peripheral and iPhone 7 is BLE Central.
iPhone's application uses `read long characteristc` and `write long characteristic` commands to operate.
Problem not encountered on previous iPhones (tested with 4s, 5, 6 and 6s).
We thought its related to MTU but its not true - reducing MTU negotiation size not helps at all.

PS: We expect any advice which can helps us workaround this problem.

SiarheiVolkau, 2017-09-13:
There are my backtrace decryption which may help figure out problem:

`function | func_start_addr | backtrace_addr`
`r_lld_evt_update_create | 0x40047cd8 | 0x40047de7`
`r_lld_con_update_req | 0x40049b60 | 0x40049b99`
`r_llc_map_update_finished | 0x40045260 | 0x400457cd`
`r_ke_stats_get | 0x4001a3f0 | 0x4001a637`
`r_ke_event_schedule | 0x40019cdc | 0x40019d11`
`r_rwip_schedule | 0x40055b38 | 0x40055b4d`
`r_rw_schedule | ----- | 0x401729ef`
`btdm_controller_task | ----- | 0x40172a3f`

hackrid, 2017-09-15:
Not familiar with the topic but I fear that the trace just tells you about a crash somewhere in the controller task.
Maybe you can debug the problem with breakpoints and stepping?

hackrid, 2017-09-15:
At least you have the PC and other values:
0x40047de7. 
What's there? Instructions? Maybe a memory access to an invalid region ;)

SiarheiVolkau, 2017-09-15:
@hackrid  Almost all backtrace functions placed in closed source library, so im not have any idea how to debug that (disassembly not helps there).

Definitely its a memory access violation.
0x40047de7 - last executed faulty instruction address.

So we need help from Guru's, who have access to source code.

hackrid, 2017-09-15:
https://www.google.com.pe/search?q=btdm_controller_task&oq=btdm_controller_task&aqs=chrome..69i57j69i60l2.1111j0j4&sourceid=chrome-mobile&ie=UTF-8

That's super sad 😔 
Why did they chose this kind of stack? 

_Sent from my Lge LG-D855 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

hackrid, 2017-09-16:
Did you check that? 
https://www.esp32.com/viewtopic.php?f=13&t=1692

TianHao-Espressif, 2017-09-25:
@SiarheiVolkau , could you tell me your ESP-IDF's GIT commit ID? I need to know the version.

SiarheiVolkau, 2017-09-25:
@TianHao-Espressif esp-idf version we use 2861f3e88ee4f08913c8ae77195d094412c9247d.

SiarheiVolkau, 2017-09-25:
Let try to explain data flow which cause this issue:

ESP32 used as GATT server for iPhone and GATT server for some other devices.

1. iPhone app sends long piece of data (Write Long Characteristic) to ESP32
2. ESP32 start scanning devices in neighborhood
3. ESP32 start connecting to founded devices
4. in the same time (2,3) iPhone periodically poll ESP32 for achieving status (Read Long Characteristic)

Yulong-espressif, 2017-09-28:
@SiarherVolkau Can you update your SDK? In the newest version of the SDK we don't detected this issue.

SiarheiVolkau, 2017-09-28:
@Yulong-espressif okay, we will try newest SDK. But i want to ask - you are reproduce problem with old SDK?

closed 2017-10-16 by FayeY