espressif/esp-idf #996: fatal error: memory: No such file or directory (arduino import issue)

sansillusion, 2017-09-13:
Hi, I am trying to import an Arduino project and I get

    fatal error: memory: No such file or directory

when it tries to compile at (wifiClient.h)
What can I do ? I still am not sure if it's possible to do but the project works great in Arduino IDE.
Arduino code here (https://github.com/sansillusion/esp32GrowSystem)

nkolban, 2017-09-14:
It might be that this is the correct repository on Github against which to report such issues ...

https://github.com/espressif/arduino-esp32

sansillusion, 2017-09-14:
Thanks, I found my proble, I had to rename the my_main.c to my_main.cpp.
But I found that after giving it all library needed and fixing code a little (more strict) My Dallas sensors where sporadically working instead of being reliable like when compiled in Arduino so I switched back to Arduino IDE.

closed 2017-09-14 by sansillusion