espressif/esp-idf #997: spiffs headers missing in repository?

enrikb, 2017-09-13:
Hi,

I just started playing around with ESP32 systems.

When I tried to build the 'blink' example, I noticed that obviously to header files are missing in the repository (branch master) at this time:

- spiffs.h
- spiffs_nucleus.h

Cheers,
Enrik

krzychb, 2017-09-14:
Hi Enrik,

Maybe your clone of ESP-IDF repository is missing spiffs submodule?
This spiffs component including submodule has been added just in last couple of days.
On GitHub it is visible as [components/spiffs/ @ 794f047](https://github.com/espressif/esp-idf/tree/dd87deb2781a1eb2d70122816d5061199b059812/components/spiffs)

If this is the case then run `git submodule update --initt` inside your local clone of ESP-IDF repository and it should be added.

enrikb, 2017-09-14:
Hi Krzysztof,

Thanks for your response.
My initial checkout / build run was just after spiffs had been added. Also, I can't see the mentioned header files at https://github.com/espressif/esp-idf/tree/dd87deb2781a1eb2d70122816d5061199b059812/components/spiffs/include

Can you please double check?

Thanks,
Enrik

krzychb, 2017-09-14:
Hi again :smile: 
The point it that this header is in another location.
It is inside the submodule,, see https://github.com/pellepl/spiffs/tree/794f0478d2aa9c978c3844da6e97f14239a1e061/src
Check if you have the whole submodule on your PC. 
If not, then use command I desribed above to get it.
Krzysztof

enrikb, 2017-09-14:
Thanks, after the submodule update --init, all files are available.

closed 2017-09-14 by enrikb