espressif/esp-idf #1001: Memory leak in commit de10032

jaracil, 2017-09-14:
I isolated a commit with memory leak.

commit: de10032717d18b323b6d49a7903167c6621e09a5

This commit is too large and I could not locate the place of the leak. There are many changes to the new memory allocation system

lucazader, 2017-09-14:
The espressif team might need some more information about what your program does and what parts of the idf you are using to try and track down the bug.

Does the leak increase after a certain function or area of your code is called?

jaracil, 2017-09-15:
My use case is a BLE device. I print the free heap after each cycle of connection / disconnection from BLE central. The heap after cycles remains stable but since the indicated commit there is a ~1k leak each cycle.

Hope that helps but checking the leak within the conn/disconn cycle is very complex, since I make intensive use of the heap.

Thanks.

lucazader, 2017-09-15:
Do you do any communication (eg and read write or notifys of any
characteristic) between the connection and disconnection?

I reported a memory leak in the set characteristic function a while back,
however I am unsure if it has been fixed yet.

jaracil, 2017-09-15:
Yes. In communications are  Write command and Notifications of one characteristic.

If you take a look at the commit, you will see that there are a lot of substitutions like: 
GKI_getbuf(...) -> osi_malloc(...)
GKI_freebuf(...) ->  osi_free(...)

My intuition tells me that there is a malloc without its free. But there are some parts of the code where it is not trivial to follow it.

garacil, 2017-09-16:
Hello everyone!

My name is German Aracil Boned. I'm from spain. It could be, but I'm not James Bond's family :-)

Is posible show a debug if you add file "allocator.h" in your project and call function osi_mem_dbg_show()" after connect event. You need modify file allocator.h first because function osi_mem_dbg_show() have a mistake like "BT: osi_mem_dbg_record invalid !! size 0". You need comment these lines like:
 
/*    if (!p || size == 0) {
        LOG_ERROR("%s invalid !! size %i\n", __func__, size);
        return;
    }
*/

Remember set "CONFIG_BLUEDROID_MEM_DEBUG" first and compile. Please if is posible, dump result here to compare with me.

I have the same problem but, only 908 bytes of memory leak every one. Remember, "you have 2 responsibilities regarding any block of memory allocated: (1) always preserve a pointer to the starting address for the block of memory so, (2) it can be freed when it is no longer needed. Freeing the memory is up to you."

First connection: Free heap: 113492
Second connection: Free heap: 112584

My dump:

First connection:

E (2787) BT: --> p 0x3ffdbeb8, s 8, f btc_to_bta_adv_data, l 262
E (2797) BT: --> p 0x3ffd2a88, s 24, f fixed_queue_new, l 42
E (2797) BT: --> p 0x3ffd2aa4, s 20, f osi_calloc_func, l 150
E (2807) BT: --> p 0x3ffd5790, s 24, f fixed_queue_new, l 42
E (2817) BT: --> p 0x3ffd5800, s 20, f osi_calloc_func, l 150
E (2817) BT: --> p 0x3ffd58c0, s 20, f osi_calloc_func, l 150
E (2827) BT: --> p 0x3ffd5abc, s 32, f osi_calloc_func, l 150
E (2837) BT: --> p 0x3ffd5ae0, s 168, f osi_calloc_func, l 150
E (2837) BT: --> p 0x3ffd5b8c, s 24, f fixed_queue_new, l 42
E (2847) BT: --> p 0x3ffd87a8, s 20, f osi_calloc_func, l 150
E (2857) BT: --> p 0x3ffd97e4, s 32, f osi_calloc_func, l 150
E (2857) BT: --> p 0x3ffd9808, s 136, f osi_calloc_func, l 150
E (2867) BT: --> p 0x3ffd98e8, s 32, f osi_calloc_func, l 150
E (2877) BT: --> p 0x3ffd990c, s 136, f osi_calloc_func, l 150
E (2877) BT: --> p 0x3ffd99ec, s 32, f osi_calloc_func, l 150
E (2887) BT: --> p 0x3ffd9a10, s 136, f osi_calloc_func, l 150
E (2887) BT: --> p 0x3ffdb1cc, s 24, f fixed_queue_new, l 42
E (2897) BT: --> p 0x3ffdb23c, s 20, f osi_calloc_func, l 150
E (2907) BT: --> p 0x3ffdb2fc, s 24, f fixed_queue_new, l 42
E (2907) BT: --> p 0x3ffdb36c, s 20, f osi_calloc_func, l 150
E (2917) BT: --> p 0x3ffdb42c, s 20, f osi_calloc_func, l 150
E (2927) BT: --> p 0x3ffdb444, s 24, f fixed_queue_new, l 42
E (2927) BT: --> p 0x3ffdb4b4, s 20, f osi_calloc_func, l 150
E (2937) BT: --> p 0x3ffdb574, s 24, f fixed_queue_new, l 42
E (2947) BT: --> p 0x3ffdb5e4, s 20, f osi_calloc_func, l 150
E (2947) BT: --> p 0x3ffdb6a4, s 24, f fixed_queue_new, l 42
E (2957) BT: --> p 0x3ffdb714, s 20, f osi_calloc_func, l 150
E (2957) BT: --> p 0x3ffdb7d4, s 24, f fixed_queue_new, l 42
E (2967) BT: --> p 0x3ffdb844, s 20, f osi_calloc_func, l 150
E (2977) BT: --> p 0x3ffdb904, s 128, f allocate_svc_db_buf, l 1340
E (2977) BT: --> p 0x3ffd0cf4, s 8, f osi_calloc_func, l 150
E (2987) BT: --> p 0x3ffdb988, s 24, f fixed_queue_new, l 42
E (2997) BT: --> p 0x3ffdb9f8, s 20, f osi_calloc_func, l 150
E (2997) BT: --> p 0x3ffdbab8, s 135, f btm_ble_adv_filter_init, l 1251
E (3007) BT: --> p 0x3ffdbb44, s 24, f fixed_queue_new, l 42
E (3017) BT: --> p 0x3ffdbbb4, s 20, f osi_calloc_func, l 150
E (3017) BT: --> p 0x3ffdbc74, s 128, f allocate_svc_db_buf, l 1340
E (3027) BT: --> p 0x3ffd2abc, s 8, f osi_calloc_func, l 150
E (3037) BT: --> p 0x3ffdbcf8, s 128, f allocate_svc_db_buf, l 1340
E (3037) BT: --> p 0x3ffd5bfc, s 8, f osi_calloc_func, l 150
E (3047) BT: --> p 0x3ffdbdd0, s 32, f osi_calloc_func, l 150
E (3057) BT: --> p 0x3ffdbdf4, s 68, f osi_calloc_func, l 150
E (3057) BT: --> p 0x3ffd5c08, s 4, f config_new_empty, l 62
E (3067) BT: --> p 0x3ffdbe90, s 20, f osi_calloc_func, l 150
E (3077) BT: --> p 0x3ffdcb1c, s 280, f btc_transfer_context, l 123
E (3077) BT: --> p 0x3ffdbef0, s 18, f btm_ble_resolving_list_init, l 951
E (3087) BT: --> p 0x3ffdbf08, s 3, f btm_ble_resolving_list_init, l 952
[... 142 lines cut ...]
E (14517) BT: --> p 0x3ffdbf6c, s 20, f osi_calloc_func, l 150
E (14517) BT: --> p 0x3ffdc34c, s 20, f osi_calloc_func, l 150
E (14527) BT: --> p 0x3ffdc40c, s 24, f fixed_queue_new, l 42
E (14527) BT: --> p 0x3ffdc47c, s 20, f osi_calloc_func, l 150
E (14537) BT: --> p 0x3ffdc5e4, s 24, f fixed_queue_new, l 42
E (14547) BT: --> p 0x3ffdc798, s 128, f allocate_svc_db_buf, l 1340
E (14547) BT: --> p 0x3ffdc694, s 8, f osi_calloc_func, l 150
E (14557) BT: --> p 0x3ffdc600, s 20, f osi_calloc_func, l 150
E (14567) BT: --> p 0x3ffdc618, s 24, f fixed_queue_new, l 42
E (14567) BT: --> p 0x3ffdc8c4, s 20, f osi_calloc_func, l 150
E (14577) BT: --> p 0x3ffdc984, s 24, f fixed_queue_new, l 42
E (14587) BT: --> p 0x3ffdc9f4, s 20, f osi_calloc_func, l 150
E (14587) BT: --> p 0x3ffdcaec, s 24, f fixed_queue_new, l 42
E (14597) BT: --> p 0x3ffdcb5c, s 20, f osi_calloc_func, l 150
E (14607) BT: --> p 0x3ffdcc1c, s 24, f fixed_queue_new, l 42
E (14607) BT: --> p 0x3ffd2b1c, s 280, f btc_transfer_context, l 123
E (14617) BT: --> p 0x3ffdcc8c, s 20, f osi_calloc_func, l 150
E (14627) BT: --> p 0x3ffdcd4c, s 28, f btc_transfer_context, l 123
E (14627) BT: --> p 0x3ffdcd6c, s 28, f btc_transfer_context, l 123
E (14637) BT: --> count 106

I'm still looking for the problem...

King regards,
   German Aracil Boned

projectgus, 2017-09-19:
There is also a new leak tracing feature in IDF which should help narrow down the place where this leak is occuring:
https://esp-idf.readthedocs.io/en/latest/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks

TianHao-Espressif, 2017-09-25:
@garacil , what's the procedure of your test? Do BLE connect to GATT-SERVER or BLE connected from  GATT_CLIENT?

jaracil, 2017-09-25:
Esp32 is GATT_SERVER.

blueMoods, 2017-10-17:
@jaracil @garacil The memory leak has been fixed, please use the latest esp_idf. Thanks.

closed 2017-10-23 by FayeY