espressif/esp-idf #1003: Spiffs examples throws error while compiling

llewellynnu, 2017-09-15:
While compiling the example for spiffs if throws the following error:

...../esp-idf/components/spiffs/./esp_spiffs.c:16:20: fatal error: spiffs.h: No such file or directory

Regards

krzychb, 2017-09-15:
SPIFFS has been added to ESP-IDF repository on 11-Sep-2017 (five days ago) with https://github.com/espressif/esp-idf/commit/dd87deb2781a1eb2d70122816d5061199b059812.
Have you updated ESP-IDF on your PC to the latest master release?

llewellynnu, 2017-09-15:
Hi, I have updated the library hence I can see the spiffs example in the example folder. 
The IDF is on latest commit

Regards

llewellynnu, 2017-09-15:
The submodule spiffs don't show up when I pull from the IDF. 

Regards

igrr, 2017-09-15:
https://github.com/espressif/esp-idf/issues/997#issuecomment-329584637

krzychb, 2017-09-15:
https://github.com/espressif/esp-idf/issues/990#issuecomment-328756507 :smile:

closed 2017-09-18 by llewellynnu