espressif/esp-idf #1004: esp_wifi_set_country doesn't change the country/region

jakkra, 2017-09-15:
I'm trying to set the country to limit the channels our device (station) is allowed to operate on.
When I call `esp_wifi_set_country(WIFI_COUNTRY_X)` it returns `ESP_OK` but I get a debug print which says 
`E (12722) wifi: wifi_set_chan_range 1722 ret=1`. Which I suspect is an error log, because of leading 'E'? 

When I call `esp_wifi_get_country(&currentEspCountry)` after setting country, it always returns `ESP_OK`, but `currentEspCountry` is always 0 (WIFI_COUNTRY_CN), which is the default according to the docs. So `esp_wifi_set_country` setting doesn't seem to apply.

Also every other time I call `esp_wifi_set_country(WIFI_COUNTRY_X)` it returns error code -1.

I have tried setting country with wlan enabled/disabled/before config, but it still doesn't apply.

There is not much documentation about `esp_wifi_set_country`, what exactly does it do?
Will it stop me from connecting to channels which are banned in the region set, will it stop me from getting scan results of APs on banned channels?

esp-idf: release/v2.1

negativekelvin, 2017-09-15:
Also reported here https://esp32.com/viewtopic.php?f=14&t=2109

liuzfesp, 2017-10-09:
Hi @jakkra API esp_wifi_set_country is rewrite in latest IDF, please help to check it.

my-abousamra, 2017-10-09:
Hi @liuzfesp I tried it right now and I got the same error

liuzfesp, 2017-10-10:
@my-abousamra could you show me how you use this API and what's your expected result please?

my-abousamra, 2017-10-10:
@liuzfesp

I use it as following:
wifi_country_t wifi_country = WIFI_COUNTRY_CN ;
ESP_ERROR_CHECK ( esp_wifi_set_country ( wifi_country ) );
I got:
E (12722) wifi: wifi_set_chan_range 1722 ret=1
I expect to execute without errors and set channels numbers according to country specified.

This test was done with the latest esp idf

liuzfesp, 2017-10-10:
@my-abousamra, I think you still use the old esp_wifi_set_country, please pull the latest IDF master branch and try it.

closed 2017-11-21 by FayeY