espressif/esp-idf #1008: Pins for using SPI >40MHz

gamer-cndg, 2017-09-16:
I'm currently writing an ILI9225 SPI driver, based on the `spi_master` example. Using the pin numbers in this example, the maximum allowed frequency is 26 MHz. Anything beyond that and it will throw an error message saying

```
spi_master: spi_bus_add_device(234): No speeds >26MHz supported for full-duplex, GPIO-matrix SPI transfers
``` 

I am configuring the bus as follows:

```
   //chip select = 22, DC=21,RST=18
    spi_bus_config_t buscfg={
        .miso_io_num=PIN_NUM_MISO, // 25 (unconnected)
        .mosi_io_num=PIN_NUM_MOSI, // 23
        .sclk_io_num=PIN_NUM_CLK, // 19
        .quadwp_io_num=-1,
        .quadhd_io_num=-1,
		.max_transfer_sz = 8 * 2 * 176 * 220
    };
    spi_device_interface_config_t devcfg={
        .clock_speed_hz=26*1000*1000,           //Clock out at 26 MHz
        .mode=0,                                //SPI mode 0
        .spics_io_num=PIN_NUM_CS,               //CS pin (22)
        .queue_size=7,                          //We want to be able to queue 7 transactions at a time
        .pre_cb=lcd_spi_pre_transfer_callback,  //Specify pre-transfer callback to handle D/C line
    };
```

The documentation at https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/spi_master.html#_CPPv216spi_bus_config_t says 

> You can use this structure to specify the GPIO pins of the bus. Normally, the driver will use the GPIO matrix to route the signals. An exception is made when all signals either can be routed through the IO_MUX or are -1. In that case, the IO_MUX is used, allowing for >40MHz speeds.

The documentation doesn't however provide an example of a possible pin configuration in which "all signals can be routed through IO_MUX" and doesn't provide references to technical manuals in which IO_MUX is explained. 

I have two requests:

- Add a reference to the technical manual (https://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf), beginning page 42 to 79. Maybe also add a new section in the documentation explaining these pin choices?
- Provide an example pin configuration so that developers don't have to dig through ~40 pages of multiplexers, pin lists and register descriptions

It would be very nice if someone with a higher technical understanding of the ESP32 could take a look at this and help me figure out a pin mapping for which the ">40MHz" speeds can be reached. 

Thank you.

negativekelvin, 2017-09-17:
Technical reference manual table 19 function 2

ginkgm, 2017-11-20:
@gamer-cndg 

1. For the operating frequency, ESP32 master clock is divided from APB clock (80MHz), so only 80, 40, 26.6, 20, 16, 13.3, 11.4, 10 MHz and lower are available.

2. The master needs a setup time of 2 APB clocks before the sample edge on the MISO line. When the GPIO mux is used, 2 APB  clocks of input delay are added to the MISO, increase 2 APB clocks to the setup time (in this case 4 APB clocks total). 

    This is the main reason we can not allow using of high frequency -- or master reads wrong data. 
    For this reason when GPIO matrix is used, only freq below 10MHz (80Mhz/2/4) works under all conditions; and when GPIO matrix is not used, only 20MHz (80MHz/2/2) works under all conditions. 
    Higher frequency communications are all achieved by some tricks, thus some restrictions exist.

3. In your case not reading, the restrictions and frequency limitations do not exist. You can remove the frequency check manually and set frequency to the list in (1). IMO ESP32 will work well in this case, no matter GPIO matrix is used or not. (we're discussing whether to add a new flag SPI_MASTER_NO_READING which disable the RX channel and checks for the RX)

    BTW, by ILI9225 spec, the maximum freq is 12.5MHz, please do it at your own risk. By our experience, speed allowed for writing is a little higher than reading.

4. About the documentation,
    1. I think it is a good idea to add IO_MUX pins into the programmer's guide
    2. About the TRM, the writing structure is (1) MUXs and PINs, (2) then comes all peripherals, without discussing the pins again. It's big change to the document, we'll discuss whether to do this.

closed 2017-12-11 by FayeY