espressif/esp-idf #1009: Unable to get an mbedTLS server to work - example requested

nkolban, 2017-09-17:
My goal is to create a listening socket (a socket server app) running on the ESP32 where the ESP32 server (listening on the socket) is using TLS/SSL to secure the connection.  To that end, I am trying to use the embedTLS APIs but having a problem.  I create a self signed certificate and private key and then have an application that looks as follows:

https://pastebin.com/rRBsd9wx

This is an almost identical copy of the mbedTLS server sample supplied by mbed:

https://tls.mbed.org/api/ssl__server_8c_source.html

The symptom is that when I hit the socket server running on the ESP32, immediately after the handshake, the connection to the peer returns EOF indicating that the partner has closed the socket.

Here is the tail of a some mbedtls trace:

```
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:6567: <= handshake
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:6743: => read
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:3721: => read record
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:2208: => fetch input
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:2366: in_left: 0, nb_want: 5
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:2390: in_left: 0, nb_want: 5
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:2391: ssl->f_recv(_timeout)() returned 0 (-0x0000)
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:3875: mbedtls_ssl_fetch_input() returned -29312 (-0x7280)
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:3729: mbedtls_ssl_read_record_layer() returned -29312 (-0x7280)
Result: [size: 0]

/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:7344: => free
/home/kolban/esp32/esptest/esp-idf/components/mbedtls/library/ssl_tls.c:7409: <= free
V (20244) sslServer: All done
```

Looking through all the examples provided by ESP-IDF, I find an excellent example for an mbedTLS SSL request (calling an external SSL server) but I find no similar example for the ESP32 being an mbedTLS SSL server.  I do find an example for being an openSSL server ... but my immediate thinking was to use mbedTLS ...

projectgus, 2017-09-18:
Hi @nkolban ,

What program are you using to connect at the moment?

If you use the openssl command line to connect, what does it output?

```
openssl s_client -debug -connect addr:port
```

nkolban, 2017-09-18:
Howdy,
I hadn't heard of that `openssl` test command.  Very nice.  I ran it and the attached trace (https://pastebin.com/F3u2XvGK) was the result.   Running that test seemed to show that the SSL handshake succeeded (but so did my original).  My ESP32 client then tries to read post handshake data and this time blocks with no data available (using openssl) but fails in a read when using a browser (because the stream is flagged as EOF).

It also appears that this issue (the one raised in this thread) has been seen by at least one other user back in June.  From examination of the thread in the forum, it appears that wasn't resolved.  See:

https://esp32.com/viewtopic.php?f=13&t=2019&p=14212

I don't claim for one moment that I have gotten my mebdTLS server correct.  The information I am working from are raw mebdTLS samples associated with mbedTLS.  I am assuming we don't have a working mbedTLS ESP32 socket server to hand?

dpgeorge, 2017-09-18:
@nkolban MicroPython running on the ESP32 uses mbedtls for secure connection (via the MicroPython `ussl` module) and both the client and server are working fine.  You may find this code useful: https://github.com/micropython/micropython-esp32/blob/esp32/extmod/modussl_mbedtls.c

projectgus, 2017-09-18:
Hi @nkolban,

If the openssl command line client can maintain a connection then it sounds like the fundamentals of the TLS server are working correctly. The problem is that the browser is closing the connection after the handshake completes rather than sending a GET request. This is either a problem at the HTTPS application layer (ie one layer above TLS), or some property of the TLS connection or certificate that the browser won't accept.

I would recommend trying to debug from the browser end - for example look for information in the "connection info" icon on the toolbar, try using the Debugging tools. Or try sending requests using a command line browser like curl or wget.

If I had to guess, I'd guess a problem with the certificate - either it has the wrong usage flags set, or the certificate CN or other metadata is invalid in some way.

Angus

nkolban, 2017-09-18:
Do we have a recipe for creating a certificate for use with a local ESP32 server?

The recipe I am using just now is:

```
openssl genrsa 2048 > private.pem
openssl req -x509 -days 1000 -new -key private.pem -subj "/C=US/ST=Texas/L=Dallas/CN=192.168.1.16" -out certificate.pem
```

projectgus, 2017-09-18:
I generated certs the same way that you have and connected to a slightly modified version of the posted code, via both Firefox & Chromium (essentially Chrome).

Both browsers warn (rightly) that the connection is insecure before continuing. In both cases, this means they close the connection immediately after the handshake and then open a new connection if the user chooses to proceed.

With the server code as-written, the server loop doesn't detect the connection is closed (read of 0 bytes) and keeps waiting to read on an already-closed socket. This might also be a bug in the IDF TCP/IP layer (maybe the second attempt to read from a closed socket should return an error), I'll check that. However I noticed both the mbedTLS server examples check for `<= 0` not `< 0`.

I was able to make the code work as-is by pressing "reset" on the ESP32 in between the initial page load and the "ignore the warning" page-load. For a better solution, here's a version which works for me in Firefox & Chrome (example-boilerplate added for WiFi connection):
https://gist.githubusercontent.com/projectgus/f30eaca579414071e0b4a84598b8fd68/raw/1a48611266ae6055d8cf593b5e80344cfbd9b2f6/gistfile1.txt

nkolban, 2017-09-18:
Thanks @projectgus 

Your tips and hints were spot on.  After some digging, I found a recipe (thank you stackoverflow) that works great for me just now.

I start Chrome with the flag "--ignore-certificate-errors".

Naturally, this is an awful thing to do for normal browser usage but for development, this is just what the doctor ordered.  I have a belief that my self signed certificate was in error.  I am reading some docs that say that "CN=" (Common Name) is no longer how Chrome identifies validity in a certificate but instead uses something called a "Subject Alternate Name (SAN)".  I hadn't been using that, just the Common Name.  I'll pursue this some more and write up what I find.  However I am now pretty comfortable that the issue is "browser and certificates" and not ESP-IDF and mbedTLS.

Thank you kind sir ...

Here are some links I found useful:

https://superuser.com/questions/27268/how-do-i-disable-the-warning-chrome-gives-if-a-security-certificate-is-not-trust
https://stackoverflow.com/questions/10175812/how-to-create-a-self-signed-certificate-with-openssl/27931596#27931596
https://support.google.com/chrome/a/answer/7391219?hl=en

closed 2017-09-18 by nkolban