espressif/esp-idf #1010: New spiffs component doesn't appear to have C++ guards

nkolban, 2017-09-18:
When I try and compile/link a C++ application with the new SPIFFS component, I end up with unresolved externals.  Debugging, it appears that the cause of this is that the spiffs/include/spiffs.h file doesn't have C++ guards contained within it.

A circumvention is to code:

```
extern "C" {
   #include <spiffs.h>
}
```

in your application code.  It is my belief that the explicit bracketing with the C specifier should not be necessary in application code.

igrr, 2017-09-19:
That's pretty odd. I have followed our submodule commit to SPIFFS repository and opened spiffs.h file. Inside, i find that guards are, in fact, present:

https://github.com/pellepl/spiffs/blob/794f0478d2aa9c978c3844da6e97f14239a1e061/src/spiffs.h#L10

Can you please see if the contents of spiffs.h file in your work directory matches the one at the link above?

nkolban, 2017-09-19:
Sorry @igrr 
My description of the problem was awful .... I transposed a file name.

The file that we want to add guards on is **not** `spiffs.h` but is instead `esp_spiffs.h`.

This file:

https://github.com/espressif/esp-idf/blob/master/components/spiffs/include/esp_spiffs.h

As such, the puzzle/problem still stands ... and I humbly apologized for pointing you to the wrong file.  I'll take what ever punishment you deem fit.

Neil

igrr, 2017-09-23:
Fixed in https://github.com/espressif/esp-idf/commit/14b12a27cf8072f38bda7c5dbdc8ed1e5c40583d, thanks for reporting!

closed 2017-09-23 by igrr