espressif/esp-idf #1012: File append not working with SPIFFS

michprev, 2017-09-18:
I am experiencing a problem when trying to open a file from SPIFFS storage in append mode using C standard file api. Read/write mode works with no problems.

me-no-dev, 2017-09-19:
can you please post some test code? I tested ```fopen(path, "a")``` and it worked as expected.

michprev, 2017-09-19:
sure, example code: [https://github.com/michprev/flyhero-esp32/tree/224f8e446d2d7854198ae4b79348077e04d56ace/Logging](https://github.com/michprev/flyhero-esp32/tree/224f8e446d2d7854198ae4b79348077e04d56ace/Logging)

Also tried to use c++ ofstream class. In this case file was not created.

nkolban, 2017-09-20:
What is the exact nature of the problem?  Is it that fopen() for append returns null?  If so, what is the value of errno immediately after the fopen()?

michprev, 2017-09-20:
I get `ENOENT` (No such file or directory) right after fopen(). Running 
```
file = fopen(path, "w");
fclose(file);

file = fopen(path, "a");
```

works as expected. It seems like file is not created (if not exists) with "a" option.

me-no-dev, 2017-09-20:
Issue found :) expect it to be resolved when changes are merged

me-no-dev, 2017-09-20:
@nkolban issue was that nonexistent file was not created when it's opened with append. I guess SPIFFS append worked in a different way ;) Fix will be in today or tomorrow :)

igrr, 2017-09-23:
Fixed in https://github.com/espressif/esp-idf/commit/14b12a27cf8072f38bda7c5dbdc8ed1e5c40583d.

closed 2017-09-23 by igrr