espressif/esp-idf #1014: error: unknown type name 'class'

hasbisevinc, 2017-09-18:
I am using ble gatt server example. I have a Square.cpp file which has Square class. When I include it from main function or other files, I get "square.h:6:1: error: unknown type name 'class'" error.

I tried to make main.c to main.cpp. But this caused lots of compile errors: " error: invalid conversion from 'int' to 'gpio_num_t' [-fpermissive]" " error: jump to case label [-fpermissive]" etc...

How can I use object oriented features with esp-idf?

negativekelvin, 2017-09-18:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/README.md

nkolban, 2017-09-20:
Maybe also:

https://www.youtube.com/watch?v=-ttiPfmrehU

hasbisevinc, 2017-09-20:
Hi nKolban,

Thank you for answering. I am trying to convert my existing project into c++. I am currently using Bluetooth GATT server codes in the project. When I include cpp_utils into components directory and change main.c to main.cpp (with extern c), I got lots of error from Bluetooth codes. 

Do I need to change my existing code, or do I need to use any other Bluetooth GATT server example?

nkolban, 2017-09-20:
I think we'll have to see a pastebin of the errors to see more context.

hasbisevinc, 2017-09-20:
Hi again,
Here is the error message: https://paste.ubuntu.com/25581083/
By the way I am not using Arduino, I am using only esp32.

nkolban, 2017-09-20:
C++ is (by default) stricter in its type checking.  From a first glance, there doesn't appear to be any ESP32 or ESP-IDF issues but rather some distinctions between what is permissible in C vs what is permissable in C++.  You will need to go through these case by case to correct.

For example:

imagine a function defined as:
```
gpio_set_direction(gpio_num_t gpioNum, gpio_mode_t mode);
```

if you tried to call this with:
```
int pin = 12;
gpio_set_direction(pin, GPIO_MODE_OUTPUT);
```
That would be a coding error.  The reason being is that the data type you supplied for the first parameter is `int` while what was expected was `gpio_num_t`.

C is more tolerant of such things and ignores them.  In C++ one would code:

```
int pin = 12;
gpio_set_direction((gpio_num_t)pin, GPIO_MODE_OUTPUT);
```
or something similar.

While C++ is **mostly** a superset of C, it is stricter and catches more logical errors than a general loose C compiler.

hasbisevinc, 2017-09-20:
Thank you very much, 
I will try to fix them step by step

closed 2017-10-03 by FayeY

arnold-b, 2018-03-24:
I looks like esp-idf don´t work with c++. Does anyone have a bigger sample which works ? And its realy tested ?

chegewara, 2018-03-24:
This library is all c++ wrapper and works fine
https://github.com/nkolban/esp32-snippets

arnold-b, 2018-03-24:
thanks i try it

arnold-b, 2018-03-25:
Now my small sample works. 

 

It will be a huge conversion from 

ESP8266 Arduino to ESP32 Eclipse. 

 

All my libraries are C++ 

 

Thanks Arnold

 

https://github.com/nkolban/esp32-snippets