espressif/esp-idf #1016: Windows pre-compiled toolchain build/flash/monitor does't work (20170918)

lucazader, 2017-09-18:
Hi All,

I am having a bit of trouble with the new version of the ESP32 percompiled toochain for windows found at [this link](https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170918.zip) (-20170918.zip) which is in [windows-setup.rst](https://github.com/espressif/esp-idf/blob/master/docs/get-started/windows-setup.rst)

I can download and unzip this fine, however there are a few problems:
1) python is not found. I have installed this using `pacman -S python` (i have also tried all the issues with `pacman -S python2`)
2) pyserial is also not installed. because PIP is also not found and I cant seem to find the `python-pip` package using pacman i use the get-pip.py script to install this and then use `pip install pyserial`

After the above steps have been done (either with python 3 or python 2) i can compile the my project. I can also flash the firmware onto my dev board fine using `make flash`.

However when I try and use make monitor (after running the `pacman -S winpty` command) i can never seem to get the monitor to work correctly. It either comes up with:
A) port already open error
B) can't use custom baud rates on this platform (using wrover kit and 2M baud)

It seems as if the the documentation in [windows-setup.rst](https://github.com/espressif/esp-idf/blob/master/docs/get-started/windows-setup.rst) is no longer sufficient to get a fully working build and debugging environment.

I will switch back to the older pre-compiled toolchain for now. but it would be good to see what other steps I need to take to get this new toolchain up and running.

Thanks in advance for any help.

CC: @projectgus

projectgus, 2017-09-19:
Hi @lucazader ,

I think the problems you're seeing are caused by running MSYS2 as "msys2.exe" rather than "mingw32.exe". If you run mingw32.exe, then it should find Python correctly and the IDF build process should work (I just re-ran the build/flash/monitor cycle on a clean Windows 10 VM and everything worked.)

The reason we need mingw32.exe is so that the version of Python is a mingw "native Windows" Python, so it can work with serial ports via native Windows APIs. The packages in MSYS2 are `mingw-w64-i686-python2` & `mingw-w64-i686-python2-pip` (the `tools/windows/windows_install_prerequisites.sh` script runs through these). Installing the `python2` package instead will install an MSYS2 (Cygwin-based) version of Python which doesn't play nicely with pyserial.

We have a couple of fixes in the pipeline to make this common (and easy to make) problem less of a hassle:

* The build system will soon warn if it detects you're running in MSYS2 or MINGW64 mode rather than MINGW32 mode.
* An installer is in development which will add a terminal link to the start menu, that starts in MINGW32 mode always.

I'm going to close this issue as I'm fairly sure this is the underlying problem, but please reopen if you think there's something different going on here.

Angus

projectgus, 2017-09-19:
(If MSYS2 mode worked on the older pre-compiled environment, I'm not entirely sure why - maybe an upstream change in MSYS2 took out the mingw paths which were set as lower priority beforehand.)

closed 2017-09-19 by projectgus

lucazader, 2017-09-19:
Thanks @projectgus 

I was running from msys32 rather than mingw32.
This did seem to work before, but not anymore.

I look forward to the installer!

Thanks for your help.

lixianyu, 2017-11-07:
lxy@lxy-PC MINGW32 /c/myProject/esp32/Rhea1_TEST
$ make
make[1]: 进入目录“/c/360yunpan/360tong_bu/Project20160127/Dione/esp-idf3/tools/kconfig”
lxdialog/check-lxdialog.sh -check cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD -lncursesw -lintl
make[1]: *** 没有规则可制作目标“/usr/include/sys/_locale.h”，由“lxdialog/checklist.o” 需求。 停止。
make[1]: 离开目录“/c/360yunpan/360tong_bu/Project20160127/Dione/esp-idf3/tools/kconfig”
make: *** 没有规则可制作目标“/c/360yunpan/360tong_bu/Project20160127/Dione/esp-idf3/tools/kconfig/conf”，由“/c/myProject/esp32/Rhea1_TEST/build/include/config/auto.conf” 需求。 停止。
who can help me?

projectgus, 2017-11-15:
@lixianyu There was a bug which is fixed in master where "make clean" did not remove all of the generated .d files from the IDF tools/kconfig directories. This can cause this error if you move IDF directories between computers or OSes.

Try `make clean`, if it doesn't work try `rm ${IDF_PATH}/tools/kconfig/*.d ${IDF_PATH}/tools/kconfig/lxdialog/*.d` (make sure to copy this command carefully as it deletes files!)