espressif/esp-idf #1017: PPPoS client Crash

llewellynnu, 2017-09-19:
Hi, 
So I updated the IDF to the latest commit as of today and also updated the tool chain. There is no change in any of my code. I am running the code on ESP32-WROVER. I get the following crash trace 

0x400d9c5c: ip4_route at /Users/llewellyn/esp/esp-idf/components/lwip/core/ipv4/ip4.c:229 (discriminator 2)

A2      : 0x3ffb6138  A3      : 0x3ffdc248  A4      : 0x00000001  A5      : 0x162af5ad  
A6      : 0x00000000  A7      : 0x00000005  A8      : 0x0000f01d  A9      : 0x3ffda550  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0xf41ea90a  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000f0d2  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x400d9c5c:0x3ffda570 0x400d9c93:0x3ffda590 0x400e1241:0x3ffda5b0 0x400df45c:0x3ffda5d0 0x400df4d9:0x3ffda610 0x400df8f0:0x3ffda630 0x400dff61:0x3ffda660 0x401115d4:0x3ffda680 0x400dece1:0x3ffda6a0
0x400d9c5c: ip4_route at /Users/llewellyn/esp/esp-idf/components/lwip/core/ipv4/ip4.c:229 (discriminator 2)

0x400d9c93: ip4_route_src at /Users/llewellyn/esp/esp-idf/components/lwip/core/ipv4/ip4.c:159

0x400e1241: udp_sendto at /Users/llewellyn/esp/esp-idf/components/lwip/core/udp.c:569 (discriminator 4)

0x400df45c: dns_send at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:787

0x400df4d9: dns_check_entry at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:984

0x400df8f0: dns_enqueue at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:1388

0x400dff61: dns_gethostbyname_addrtype at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:1493

0x401115d4: lwip_netconn_do_gethostbyname at /Users/llewellyn/esp/esp-idf/components/lwip/api/api_msg.c:1917

0x400dece1: tcpip_thread at /Users/llewellyn/esp/esp-idf/components/lwip/api/tcpip.c:474

If I revert to dd87deb2 everything works fine.
This issue is with or without SPIRAM enabled

Regards

llewellynnu, 2017-09-20:
After Further Investigation. It appears to crash after PPPoS client connection has been established and I try to Connect to AWSIoT

Regards

loboris, 2017-09-22:
I've found the temporary solution.
The cause of the crash is the function **ip4_route(const ip4_addr_t *dest)** in in _esp-idf/components/lwip/core/ipv4/ipv4.c_ (line 173).
I've pached it with some code from previous version and it works.
The patch is as follows:
```
--- ip4.c	2017-09-22 11:50:52.000000000 +0200
+++ ip4.c.patched	2017-09-22 11:40:28.000000000 +0200
@@ -172,6 +172,12 @@
 struct netif *
 ip4_route(const ip4_addr_t *dest)
 {
+#if ESP_LWIP
+  struct netif *non_default_netif = NULL;
+#if LWIP_HAVE_LOOPIF
+  struct netif *loop_default_netif = netif_find("lo0");
+#endif
+#endif
   struct netif *netif;
 
 #if LWIP_MULTICAST_TX_OPTIONS
@@ -195,8 +201,22 @@
         /* return netif on which to forward IP packet */
         return netif;
       }
+      if (netif != netif_default){
+#if LWIP_HAVE_LOOPIF
+          non_default_netif = (netif == loop_default_netif) ? NULL : netif;
+#else
+          non_default_netif = netif;
+#endif
+      }
     }
   }
+
+#if ESP_LWIP
+  if (non_default_netif && !ip4_addr_isbroadcast(dest, non_default_netif)){
+    return non_default_netif;
+  }
+#endif
+
 #if LWIP_NETIF_LOOPBACK && !LWIP_HAVE_LOOPIF
   /* loopif is disabled, looopback traffic is passed through any netif */
   if (ip4_addr_isloopback(dest)) {

```
You can save it to file (for example ip4.c.patch) and patch the original ip4.c:
`patch ip4.c ip4.c.patch`
I have to check if there are any consequences on other lwip operations.

llewellynnu, 2017-09-22:
HI, 

Thanks.

Ill have a go at it and let you know. 

Regards

loboris, 2017-10-02:
A different patch could be used wich shouldn't affect APSTA mode:
```
--- ip4.c	2017-09-22 11:50:52.000000000 +0200
+++ ip4.c.patched	2017-10-02 11:19:35.000000000 +0200
@@ -190,6 +190,10 @@
         /* return netif on which to forward IP packet */
         return netif;
       }
+      if (netif_ip4_gw(netif)->addr == 1077952522) {
+        /* return netif on which to forward IP packet */
+        return netif;
+      }
       /* gateway matches on a non broadcast interface? (i.e. peer in a point to point interface) */
       if (((netif->flags & NETIF_FLAG_BROADCAST) == 0) && ip4_addr_cmp(dest, netif_ip4_gw(netif))) {
         /* return netif on which to forward IP packet */
@@ -197,6 +201,7 @@
       }
     }
   }
+
 #if LWIP_NETIF_LOOPBACK && !LWIP_HAVE_LOOPIF
   /* loopif is disabled, looopback traffic is passed through any netif */
   if (ip4_addr_isloopback(dest)) {
```
In PPPoS gw address is set to 10.64.64.64, we can check it in ip4_route function (ip4.c) and return the netif if other checks fail.

liuzfesp, 2017-10-09:
Hi @llewellynnu could you help to try the latest IDF without your patch since we have some modifications for ip4_route.

llewellynnu, 2017-10-09:
Hi @liuzfesp 
Give me a couple of hours. I will test it and post the results here.

Regards

llewellynnu, 2017-10-09:
Hi @liuzfesp 

The client crashes without the workaround 

Here is the backtrace of the crash.

Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400da6e4  PS      : 0x00060930  A0      : 0x800da70e  A1      : 0x3ffe32b0  
0x400da6e4: ip4_route at /Users/llewellyn/esp/esp-idf/components/lwip/core/ipv4/ip4.c:229

A2      : 0x0000f01d  A3      : 0x3fffbc4c  A4      : 0x00000001  A5      : 0x3fff8f94  
A6      : 0x0000002a  A7      : 0x3fff3fa0  A8      : 0x00000000  A9      : 0x00000005  
A10     : 0x00000005  A11     : 0x00000005  A12     : 0x00000002  A13     : 0xffffffff  
A14     : 0x00001800  A15     : 0x3ffaff98  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000f0d2  LBEG    : 0x40095a20  LEND    : 0x40095a33  LCOUNT  : 0x00000000  
0x40095a20: __memcpy_aux at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:73

0x40095a33: __memcpy_aux at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:88

Backtrace: 0x400da6e4:0x3ffe32b0 0x400da70b:0x3ffe32d0 0x400e0c4d:0x3ffe32f0 0x400df499:0x3ffe3310 0x400df5a7:0x3ffe3350 0x400dfced:0x3ffe3370 0x401562a1:0x3ffe33b0 0x400deedf:0x3ffe33d0
0x400da6e4: ip4_route at /Users/llewellyn/esp/esp-idf/components/lwip/core/ipv4/ip4.c:229

0x400da70b: ip4_route_src at /Users/llewellyn/esp/esp-idf/components/lwip/core/ipv4/ip4.c:159

0x400e0c4d: udp_sendto at /Users/llewellyn/esp/esp-idf/components/lwip/core/udp.c:569 (discriminator 4)

0x400df499: dns_send at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:787

0x400df5a7: dns_check_entry at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:1014

0x400dfced: dns_enqueue at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:1388
 (inlined by) dns_gethostbyname_addrtype at /Users/llewellyn/esp/esp-idf/components/lwip/core/dns.c:1493

0x401562a1: lwip_netconn_do_gethostbyname at /Users/llewellyn/esp/esp-idf/components/lwip/api/api_msg.c:1917

0x400deedf: tcpip_thread at /Users/llewellyn/esp/esp-idf/components/lwip/api/tcpip.c:474

Regards

liuzfesp, 2017-10-09:
Thanks @llewellynnu, we will investigate it.

liuzfesp, 2017-10-10:
Hi @llewellynnu could you help to provide the xxx.elf file which generates the exception?

llewellynnu, 2017-10-10:
Hi @liuzfesp If you can provide me with your email ID I can send you the .elf file. As the firmware connects to my production servers.

Dmitry4Bh, 2017-10-11:
Hi! Try this patch https://github.com/espressif/esp-idf/pull/1100

llewellynnu, 2017-10-12:
hi @Dmitry4Bh 

It works perfectly now. Thanks.

Regards

igrr, 2017-11-14:
#1100 has been merged, closing this issue as fixed.

closed 2017-11-14 by igrr