espressif/esp-idf #1018: AP mode doesn't work

kimdowan84, 2017-09-20:
Dear MR.

I do not know from what version the wifi is not working correctly, 
but found that v3.0-dev-710 is not work at AP mode.

The sysdrome is that client can't get ip address.
Reference [TW#14352]

Thank you.

tetious, 2017-09-22:
I'm also seeing this in current master, when using SOFTAP+STA. I think the DHCP server is no longer working.

This was not a problem at commit 969f1bb.

tetious, 2017-09-24:
This is almost certainly related to DHCP. If I use a static IP matching the usual DHCP configuration, it works fine. 

I use 
```
IP: 192.168.4.2
Netmask: 255.255.255.0
Gateway 192.168.4.1
```

liuzfesp, 2017-10-09:
HI @tetious this issue already fixed in IDF master branch, please check it.

kimdowan84, 2017-10-14:
Yes, 806 version is OK!

closed 2017-10-16 by FayeY