espressif/esp-idf #1019: RMT with TX loop enabled results in crash

DavidAntliff, 2017-09-20:
When using RMT to generate a simple square wave, the application will crash if the `rmt_config_t.tx_config.loop_en` field is set to `true` and `rmt_write_items()` is called. If set to `false` it does not crash.

```
void rmt_task(void * pvParameter)
{
    rmt_config_t rmt_tx = {
        .rmt_mode = RMT_MODE_TX,
        .channel = RMT_CHANNEL_0,
        .gpio_num = GPIO_NUM_4,
        .mem_block_num = 1,
        .clk_div = 80,
        .tx_config.loop_en = true,  // <-- set this to 'true' for crash
        .tx_config.carrier_en = false,
        .tx_config.idle_level = RMT_IDLE_LEVEL_LOW,
        .tx_config.idle_output_en = true,
    };
    rmt_config(&rmt_tx);
    rmt_driver_install(rmt_tx.channel, 0, 0);

    // simple sequence of high for one clock, low for one clock
    rmt_item32_t items[2] = {0};
    items[0].level0 = 1;
    items[0].duration0 = 1;
    items[0].level1 = 0;
    items[0].duration1 = 1;

    // terminate sequence with a zero-duration item
    items[1].level0 = 0;
    items[1].duration0 = 0;

    // Note: I've also tried this with a single item - same result
    rmt_write_items(RMT_TX_CHANNEL, items, 2, false);

    while(1)
        ;
}

void app_main()
{
    xTaskCreate(&rmt_task, "rmt_task", 2048, NULL, 5, NULL);
}
```

Sometimes when it crashes I see log messages such as:

`W (90) boot: PRO CPU has been reset by WDT.`

or

`***ERROR*** A stack overflow in task ???D??@(?? has been detected.`

Other times it just restarts with no error.

100% reproducible with my setup. I'm using ESP-IDF 27574a31e7c41f6e22e10e9b9d7713c6d49f795d

DavidAntliff, 2017-09-20:
Using a longer sequence such as:

```
    rmt_item32_t items[65] = {0};
    for (size_t i = 0; i < 64; ++i)
    {
        items[i].level0 = 1;
        items[i].duration0 = 1;
        items[i].level1 = 0;
        items[i].duration1 = 1;
    }
    items[64].level0 = 0;
    items[64].duration0 = 0;
    rmt_write_items(RMT_TX_CHANNEL, items, 65, false);
```

Does not seem to result in a crash, however the signal is only seen (via oscilloscope) to execute once, after which the GPIO is idle.

DavidAntliff, 2017-09-20:
Strangely, if I set the number of items to 10 (9 level/duration items plus one terminating item) then it loops correctly and the application does not crash. But if I put this in the final while loop, I get a stack overflow: `***ERROR*** A stack overflow in task rmt_task has been detected.`

```
    while(1)
    {
        ESP_LOGI("rmt", "wait");
        vTaskDelay(10);
    }
```

Yet this does not:

```
    while(1)
    {
        vTaskDelay(10);
    }
```

DavidAntliff, 2017-09-20:
With the following code, setting `num_items` to 63 results in proper looping, but a value of 64 does not loop (items only execute once):

```
void rmt_task(void * pvParameter)
{
    rmt_config_t rmt_tx = {
        .rmt_mode = RMT_MODE_TX,
        .channel = RMT_CHANNEL_0,
        .gpio_num = GPIO_NUM_4,
        .mem_block_num = 1,
        .clk_div = 80,
        .tx_config.loop_en = true,
        .tx_config.carrier_en = false,
        .tx_config.idle_level = RMT_IDLE_LEVEL_LOW,
        .tx_config.idle_output_en = true,
    };
    rmt_config(&rmt_tx);
    rmt_driver_install(rmt_tx.channel, 0, 0);

    int num_items = 63;
    rmt_item32_t items[64] = {0};
    for (size_t i = 0; i < num_items; ++i)
    {
        items[i].level0 = 1;
        items[i].duration0 = 1;
        items[i].level1 = 0;
        items[i].duration1 = 1;
    }

    // modify the last item to terminate (adds one clock at idle level)
    items[num_items - 1].level1 = 0;
    items[num_items - 1].duration1 = 0;

    rmt_write_items(RMT_CHANNEL_0, items, num_items, false);

    while(1)
    {
        //ESP_LOGI(TAG, "wait");  // <- causes a crash for num_items <= 63
        vTaskDelay(10);
    }
}
```

Uncommenting the ESP_LOGI line will cause a crash for values of `num_items` 63 or smaller, but not for 64.

DavidAntliff, 2017-09-20:
Setting both `duration0` and `duration1` to 100 doesn't change the non-looping behaviour for 64 items. It's not crashing though either.

negativekelvin, 2017-09-20:
Make your stack bigger or put your items buffer in heap

DavidAntliff, 2017-09-21:
Heap allocation has made the crash go away - I can now use just two items, with an `ESP_LOGI()` call, and it doesn't crash - thank for for the suggestion.

```
void rmt_task(void * pvParameter)
{
    rmt_config_t rmt_tx = {
        .rmt_mode = RMT_MODE_TX,
        .channel = RMT_CHANNEL_0,
        .gpio_num = GPIO_NUM_4,
        .mem_block_num = 1,
        .clk_div = 80,
        .tx_config.loop_en = true,
        .tx_config.carrier_en = false,
        .tx_config.idle_level = RMT_IDLE_LEVEL_LOW,
        .tx_config.idle_output_en = true,
    };
    rmt_config(&rmt_tx);
    rmt_driver_install(rmt_tx.channel, 0, 0);

    size_t num_items = 64;
    rmt_item32_t * items = calloc(num_items, sizeof(*items));
    for (size_t i = 0; i < num_items; ++i)
    {
        items[i].level0 = 1;
        items[i].duration0 = 1;
        items[i].level1 = 0;
        items[i].duration1 = 1;
    }

    // terminate sequence with a zero duration item
    items[num_items - 1].level1 = 0;
    items[num_items - 1].duration1 = 0;

    rmt_write_items(RMT_CHANNEL_0, items, num_items, false);

    while(1)
    {
        ESP_LOGI(TAG, "wait");
        vTaskDelay(10);
    }
}
```

However, it still doesn't loop when the number of items is equal to or greater than 64. I've only tried a few different values 64 and greater and none of them loop. 63 and all values less seem to loop OK.

PerMalmberg, 2017-09-21:
You need to call rmt_set_tx_loop_mode() to set the loop mode.

DavidAntliff, 2017-09-21:
Is that required? The loop mode is already set to true in the config struct, passed to `rmt_config()`. Other API functions don't need to be called explicitly if already set in the config struct - so why this one? And how come looping works fine up to 63 items without that call? If what you say is true then the API is very broken.

PerMalmberg, 2017-09-21:
Oh, right. No, then it shouldn't be needed.

negativekelvin, 2017-09-22:
It appears that the driver does not support continuous mode for signals that don't fit in the channel's reserved memory block(s).  So you can try to increase the number of blocks for that channel.

DavidAntliff, 2017-09-22:
Ok, that makes sense since the 512 words are split into eight 64 word blocks. I'll give that a try.

DavidAntliff, 2017-09-22:
I haven't tried it yet, but it does seem a bit strange to me that the looping fails for 64 items, not 65 - isn't the 64th item still in the same memory block?

negativekelvin, 2017-09-22:
Yes the driver reserves a space for the zero duration item but in the edge case this should not be required.

DavidAntliff, 2017-09-22:
Ok, in my example I was providing the zero duration item myself - so perhaps it's not required?

One thing I noticed is that the is a one clock cycle delay between the last edge of the last item in the pattern and the first edge of the first item in the pattern (i.e. when it loops). I thought that it was because I was explicitly including a zero duration item in the "loop" items, and that if I omitted the zero duration item and just used the "num items" to mark the end of the sequence that I would get a "perfect" loop, but based on what you've just said, I wonder now if that won't be the case? Does the driver add the zero duration item always? I'll try it out (and examine the IDF code too) when I next get a chance.

DavidAntliff, 2017-09-25:
When 64 items fails to loop, increasing `rmt_config_t.mem_block_num` to 2 causes it to loop again. Then 127 items also loops, but 128 does not. So it looks like the memory block size does need to be consistent with the number of items (which is sensible).

Unfortunately when using only 4 items, with a low clock divider (e.g. 4), the system seems unstable and randomly reboots with a watchdog reset. I guess it's generating RMT-tx-complete interrupts faster than it can handle them.

DavidAntliff, 2017-09-25:
I'll close this because I think the original behaviours have been explained.

closed 2017-09-25 by DavidAntliff