espressif/esp-idf #1020: TCP incorrect checksums

dobegor, 2017-09-20:
I have a ESP32 and a VPS in a cloud.
Connectivity via ESP32 -> ngrok (proxy) -> VPS works perfectly.
Direct connect (without proxy) works on any machine except ESP32, resulting in error 113 (ECONNRESET).

tcpdump log: https://pastebin.com/RJERT1T5

It appears that ESP32 has huge window size and mss for some reason and almost all checksums are incorrect. It doesn't happen in case connecting via proxy.

Connecting from another device on the same Wi-Fi network works perfectly.

projectgus, 2017-09-20:
Hi,

I ran this packet capture on my router from an ESP32 earlier today, using the HTTP example modified as you mention:
[tcp_problem.pcap.zip](https://github.com/espressif/esp-idf/files/1317125/tcp_problem.pcap.zip)

The outward symptoms (that HTTP server responds to normal requests but not to requests from the ESP32) is the same for me as for you.

The TCP SYN packets, including IP checksums, look correct in the capture I took. Usually "invalid checksum" for a local packet capture is because the OS is offloading checksumming to some lower layer.

The MSS & window size look correct to me as well, and certainly not out of spec for TCP. These values can be configured in "menuconfig" under Component Config -> LWIP -> TCP. The allowable ranges are unfortunately small because it's not possible to disable IPV6 support, which means it's quite large.

However, there are some differences (TCP options, etc) between the TCP SYN packets sent by an ESP32 compared to most "big" OSes.

Do you have control of the server at 51.15.34.30? If you run a tcpdump on that server, do you see anything coming to/from the ESP32?

projectgus, 2017-09-20:
(Oops, you can ignore the ICMP packets in the pcap I attached. 192.168.66.192 is the ESP32.)

dobegor, 2017-09-20:
@projectgus tcpdump I mentioned in the first comment was taken from the problematic server. Yes, I have an access to that server.

projectgus, 2017-09-20:
>  Usually "invalid checksum" for a local packet capture is because the OS is offloading checksumming to some lower layer

If the capture is on the server, this may be worth further consideration (assuming the checksum is OK for working requests). Maybe a router somewhere is mangling the SYN packets on their way into the VPS.

dobegor, 2017-09-21:
@projectgus Unfortunately, I cannot reproduce it anymore. Apparently, hosting provider fixed their router settings or somewhat.

projectgus, 2017-09-21:
Good to hear it's working now, @dobegor . I'm going to close this issue on the assumption this wasn't an ESP-IDF problem, but please reopen if you think that's incorrect.

closed 2017-09-21 by projectgus