espressif/esp-idf #1021: wifi: bcn_timout,ap_probe_send_start

jasslo, 2017-09-20:
Hi,
I have on my ESP32 simple web server, which get data from esp8266 thru get requests and display it on web page. In future ESP32 will send this data to thingspeak or something like that. Everything works fine, but after few minutes it crash and display on monitor:

```
I (1009053) wifi: bcn_timout,ap_probe_send_start
I (1009553) wifi: bcn_timout,ap_probe_send_start
I (1010053) wifi: bcn_timout,ap_probe_send_start
I (1010553) wifi: bcn_timout,ap_probe_send_start
I (1011053) wifi: bcn_timout,ap_probe_send_start
I (1011553) wifi: bcn_timout,ap_probe_send_start
I (1012053) wifi: bcn_timout,ap_probe_send_start
I (1012553) wifi: bcn_timout,ap_probe_send_start
I (1013053) wifi: bcn_timout,ap_probe_send_start
I (1013553) wifi: bcn_timout,ap_probe_send_start
I (1014053) wifi: bcn_timout,ap_probe_send_start
I (1014563) wifi: bcn_timout,ap_probe_send_start
I (1015063) wifi: bcn_timout,ap_probe_send_start
I (1015563) wifi: bcn_timout,ap_probe_send_start
I (1016063) wifi: bcn_timout,ap_probe_send_start
I (1016563) wifi: bcn_timout,ap_probe_send_start
I (1017063) wifi: bcn_timout,ap_probe_send_start
I (1017563) wifi: bcn_timout,ap_probe_send_start
I (1018063) wifi: bcn_timout,ap_probe_send_start
I (1018563) wifi: bcn_timout,ap_probe_send_start
I (1019063) wifi: bcn_timout,ap_probe_send_start
I (1019563) wifi: bcn_timout,ap_probe_send_start
I (1020063) wifi: bcn_timout,ap_probe_send_start
I (1020563) wifi: bcn_timout,ap_probe_send_start
I (1021063) wifi: bcn_timout,ap_probe_send_start
I (1021573) wifi: bcn_timout,ap_probe_send_start
I (1022073) wifi: bcn_timout,ap_probe_send_start
I (1022573) wifi: bcn_timout,ap_probe_send_start
I (1023073) wifi: bcn_timout,ap_probe_send_start
I (1023573) wifi: bcn_timout,ap_probe_send_start
I (1024073) wifi: bcn_timout,ap_probe_send_start
W (1024333) wifi: inactive timer: now=f70b786 last_rx_time=ad73eb97 diff=50457
I (1024333) wifi: station: 00:c2:c6:0e:5f:79 leave, AID = 1
I (1024333) wifi: n:11 0, o:11 0, ap:11 2, sta:11 0, prof:11
I (1024333) app_main: station:00:c2:c6:0e:5f:79leave,AID=1

I (1024573) wifi: bcn_timout,ap_probe_send_start
I (1025073) wifi: bcn_timout,ap_probe_send_start
I (1025573) wifi: bcn_timout,ap_probe_send_start
I (1026073) wifi: bcn_timout,ap_probe_send_start
I (1026573) wifi: bcn_timout,ap_probe_send_start
I (1027073) wifi: bcn_timout,ap_probe_send_start
I (1027573) wifi: bcn_timout,ap_probe_send_start
I (1028073) wifi: bcn_timout,ap_probe_send_start
I (1028573) wifi: bcn_timout,ap_probe_send_start
I (1029073) wifi: bcn_timout,ap_probe_send_start
I (1029583) wifi: bcn_timout,ap_probe_send_start
I (1030083) wifi: bcn_timout,ap_probe_send_start
I (1030583) wifi: bcn_timout,ap_probe_send_start
I (1031083) wifi: bcn_timout,ap_probe_send_start
I (1031583) wifi: bcn_timout,ap_probe_send_start
I (1032083) wifi: bcn_timout,ap_probe_send_start
I (1032583) wifi: bcn_timout,ap_probe_send_start
I (1033083) wifi: bcn_timout,ap_probe_send_start
I (1033583) wifi: bcn_timout,ap_probe_send_start
I (1034083) wifi: bcn_timout,ap_probe_send_start
I (1034583) wifi: bcn_timout,ap_probe_send_start
I (1035083) wifi: bcn_timout,ap_probe_send_start
I (1035583) wifi: bcn_timout,ap_probe_send_start
I (1036083) wifi: bcn_timout,ap_probe_send_start
I (1036583) wifi: bcn_timout,ap_probe_send_start
I (1037093) wifi: bcn_timout,ap_probe_send_start
I (1037593) wifi: bcn_timout,ap_probe_send_start
I (1038093) wifi: bcn_timout,ap_probe_send_start
I (1038593) wifi: bcn_timout,ap_probe_send_start
I (1039093) wifi: bcn_timout,ap_probe_send_start
I (1039593) wifi: bcn_timout,ap_probe_send_start
I (1040093) wifi: bcn_timout,ap_probe_send_start
I (1040593) wifi: bcn_timout,ap_probe_send_start
I (1041093) wifi: bcn_timout,ap_probe_send_start
I (1041593) wifi: bcn_timout,ap_probe_send_start
I (1042093) wifi: bcn_timout,ap_probe_send_start
I (1042593) wifi: bcn_timout,ap_probe_send_start
I (1043093) wifi: bcn_timout,ap_probe_send_start
I (1043593) wifi: bcn_timout,ap_probe_send_start
I (1044093) wifi: bcn_timout,ap_probe_send_start
I (1044603) wifi: bcn_timout,ap_probe_send_start
```
This information will be displaying over and over.
I must mention that in the beginning wifi is in AP mode and we can configure sta ssid and password and then wifi is switched to ap+sta mode.
Someone maybe have idea what is wrong ?

jakkra, 2017-09-21:
Possibly related to https://github.com/espressif/esp-idf/issues/974

DigKleppe, 2017-11-11:
Any news yet?
I am facing the same issue. Also see my post [https://esp32.com/viewtopic.php?f=2&t=3522](url)
After 2615 messages:
2613
2614
2615
I (59562) wifi: bcn_timout,ap_probe_send_start
I (62062) wifi: ap_probe_send over, resett wifi status to disassoc
I (62062) wifi: state: run -> init (1)
I (62062) wifi: pm stop, total sleep time: 0/57061570

I (62062) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
D (62072) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Klepnet, ssid_len:7, bssid:fc:c8:97:8b:bc:ba, reason:200
V (62082) event: enter default callback
D (62092) tcpip_adapter: check: local, if=0 fn=0x40116520

D (62092) tcpip_adapter: if0 start ip lost tmr: enter
D (62092) tcpip_adapter: if0 start ip lost tmr: interval=120
D (62102) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (62112) tcpip_adapter: check: remote, if=0 fn=0x40116520

V (62112) event: exit default callback
D (64522) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Klepnet, ssid_len:7, bssid:00:00:00:00:00:00, reason:201
V (64522) event: enter default callback
D (64522) tcpip_adapter: check: local, if=0 fn=0x40116520

And so on.

Also reconnect  does not work

ricksondpenha, 2017-11-29:
Yes i get this error too!, I use aws iot mqtt pub sub and a tcp server, when too many messages are sent, wifi timeout and refuses to connect again, tried many things like disconnecting and connecting again, but only way is to restart the esp module. Any workaround for this?

0BackDoor1, 2017-12-19:
Hi,
I'm facing same issue, any news about it?

```
I (10030333) wifi: bcn_timout,ap_probe_send_start
I (10032837) wifi: ap_probe_send over, resett wifi status to disassoc
I (10032838) wifi: state: run -> init (1)
I (10032840) wifi: pm stop, total sleep time: 0/1437653599
I (10032840) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
D (10032849) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:abcdefg Guest, ssid_len:13, bssid:e0:b7:e5:b6:b7:bf, reason:200
D (10032863) tcpip_adapter: check: local, if=0 fn=0x4011df00
D (10032874) tcpip_adapter: if0 start ip lost tmr: enter
D (10032879) tcpip_adapter: if0 start ip lost tmr: interval=120
D (10032884) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (10032890) tcpip_adapter: check: remote, if=0 fn=0x4011df00

D (10035311) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:abcdefg Guest, ssid_len:13, bssid:00:00:00:00:00:00, reason:201
D (10035312) tcpip_adapter: check: local, if=0 fn=0x4011df00
D (10035317) tcpip_adapter: if0 start ip lost tmr: enter
D (10035322) tcpip_adapter: if0 start ip lost tmr: already started
D (10035328) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (10035334) tcpip_adapter: check: remote, if=0 fn=0x4011df00

D (10037749) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:abcdefg Guest, ssid_len:13, bssid:00:00:00:00:00:00, reason:201
D (10037750) tcpip_adapter: check: local, if=0 fn=0x4011df00
D (10037755) tcpip_adapter: if0 start ip lost tmr: enter
D (10037760) tcpip_adapter: if0 start ip lost tmr: already started
D (10037766) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (10037772) tcpip_adapter: check: remote, if=0 fn=0x4011df00
```

ESP-IDF branch: release/v3.0 updated at commit 0ebae99ab072fb91fd5223869d786f319db2e274 (17 dec 2017)

DigKleppe, 2017-12-19:
Solved , see post:
[www.esp32.com/viewtopic.php?f=13&t=3451&p=17503&hilit=Dig+Kleppe#p16072](url)
Memory leak in netconn new

jack0c, 2018-01-03:
@jasslo @ricksondpenha things you can do to check what's the problem:
1.check the version, give the commit ID here.
2.dump your NVS data to a file, give the file here.
3.erase the flash and flash again.
4. check menuconfig about Components config->PHY->Do phy calibration ..., you can do phy calibration every time.

0BackDoor1, 2018-01-24:
Hi,
any news about this issue?
I tried erasing the flash and with PHY calibration active, but still getting it

jack0c, 2018-01-24:
@0BackDoor1 Can you capture the packets in the air？

0BackDoor1, 2018-02-16:
@jack0c my issue doesn't occur anymore with IDF updated to latest commit available, good!

jack0c, 2018-03-22:
@0BackDoor1 Can you give commit you use. Maybe can help us and other developers. Thanks.

mzimmers, 2018-04-03:
I'm experiencing this as well. From reading this thread, and the one referenced above, it's still not clear to me whether this has been fixed, and whether it requires a change to user code. Can someone elucidate? Thanks...

0BackDoor1, 2018-05-14:
@jack0c sorry for delay, I missed email notification about this, my repo was at commit cf2600adefe7ea2f805b4c9d1e28efc4a6f23064

NOA-Vernon, 2018-06-21:
Is there any update? I also experience this issue

118979) wifi: bcn_timout,ap_probe_send_start
I (121479) wifi: ap_probe_send over, resett wifi status to disassoc
I (121479) wifi: state: run -> init (1)
I (121479) wifi: pm stop, total sleep time: 0 us / 111197816 us

I (121489) wifi: n:7 0, o:7 0, ap:255 255, sta:7 0, prof:1

0BackDoor1, 2018-06-21:
@NOA-Vernon first of all ensure to have IDF updated, the same issue occurs if the free heap size (esp_get_free_heap_size) is too low, don't know if this could help on your case

NOA-Vernon, 2018-06-21:
@0BackDoor1  thank you very much for the response,  the IDF I used is ESP-IDF v3.1-dev-961-ga255622-dirty, the free heap size is around 50000, this issue still happened again and not have a solution yet. hoping there is anyone who can elucidate or point out the problems.

bill1davis, 2018-08-02:
This is reproducible easily using the power_save example source with the current esp-idf.  I have seen it on two different mainstream routers using 100ms beacon interval and DTIM settings of 1 or 3.  I feel like it happens much more often with DTIM=1.  With DTIM=3 I have seen it run for 1/2 hour to 1 hour before getting bcn_timout and disconnecting.  I am still testing trying to work around the disconnect.  It seems like wifi gets messed up after and can't reconnect.

snahmad, 2018-09-05:
Have anyone resolve this issue. I am facing same. I think it is due to interval ram less available. but with external RAM. I get wifi alloc error.

bill1davis, 2018-09-05:
I was able to work around it by running both cores at 266 mHz. I suspect it is wake timing related.  Not my plan for the long term. I hope it gets looked at.

snahmad, 2018-09-05:
Yes I am only using one core. I think two cores with external RAM is not stable and safe on Esp32.

I am using Http Web server with SD card.

I am getting similar error
wifi: alloc eb len=192 type=2 fail

I am using WIFI with external RAM.

Internal Heap free: 41120
Heap free: 4195984:4233744

Plenty of external ram and some internal RAM.

Does in default settings WIFI only uses internal ram.

vonnieda, 2018-09-05:
@snahmad If you are using TLS and have 41k internal free, you will likely run out of memory due to TLS connections. Read through the comments on https://github.com/espressif/esp-idf/issues/2184 to see how you can switch to using external RAM for TLS, or see if you can free up more internal RAM. In my tests it seems like each TLS connection takes > 20kB of internal RAM.

snahmad, 2018-09-05:
I am not using TSL yet. in future may be.
Only unsecure HTTP Mongoose Web server and few TCP and UDP sockets connections.
I will configure to wifi use only static buffer only internal RAM.
My sector size for SD card is 32k. I can try reducing it.

snahmad, 2018-09-05:
Why WIFI is using external RAM.

I need to specify for both Tx and RX static buffer on internal RAM only for Wifi.

W (6548) wifi: m f probe req l=0

W (6699) wifi: alloc eb len=76 type=2 fail, heap:4192980

W (6700) wifi: m f probe req l=0

Stack free: 6396
Internal Heap free: 39416
Heap free: 4195048:4232808

dalerks, 2019-02-28:
Any progress I am having this issue too

snahmad, 2019-02-28:
You mean run-out internal ram. you can specify to use only static ram for Wifi in menuonfig.

liuzfesp, 2019-03-04:
@dalerks, what's your issue?  beacon timeout or the log "m f probe req l=0" is displayed? Could you paste your error log here?

dalerks, 2019-03-04:
W (351669) wifi: alloc eb len=24 type=3 fail, heap:88

W (351669) wifi: m f null

I (357609) wifi: bcn_timout,ap_probe_send_start
W (357609) wifi: alloc eb len=76 type=2 fail, heap:88

W (357609) wifi: m f probe req l=0

W (358109) wifi: alloc eb len=76 type=2 fail, heap:88

W (358109) wifi: m f probe req l=0

W (358609) wifi: alloc eb len=76 type=2 fail, heap:88

W (358609) wifi: m f probe req l=0

W (359109) wifi: alloc eb len=76 type=2 fail, heap:88

W (359109) wifi: m f probe req l=0

W (359609) wifi: alloc eb len=76 type=2 fail, heap:88

W (359609) wifi: m f probe req l=0

I (360109) wifi: ap_probe_send over, resett wifi status to disassoc
I (360109) wifi: state: run -> init (c800)
W (360109) wifi: alloc eb len=28 type=3 fail, heap:88

W (360119) wifi: m f disassoc

I (360119) wifi: pm stop, total sleep time: 290332841 us / 355180847 us

I (360129) wifi: new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
I (360129) AirCylcer: Event called
I (360259) wifi: new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
I (360259) wifi: state: init -> auth (b0)
I (360259) wifi: state: auth -> init (8a0)
I (360259) wifi: new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
I (360269) AirCylcer: Event called
Fan on for 0 
Runtime 6 
Saved time 20 
Rem time 20 
I (362319) AirCylcer: Event called
I (362449) wifi: new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
I (362449) wifi: state: init -> auth (b0)
I (362449) wifi: state: auth -> assoc (0)
I (362459) wifi: state: assoc -> run (10)
I (362489) wifi: connected with AirLinksys24G, channel 10, bssid = 60:38:e0:99:9d:35
I (362489) wifi: pm start, type: 1

I (362489) AirCylcer: Event called
I (363299) event: sta ip: 192.168.1.149, mask: 255.255.255.0, gw: 192.168.1.1
I (363299) AirCylcer: Event called
E (370139) aws_iot: failed! mbedtls_ssl_write returned -0x4e

liuzfesp, 2019-03-05:
HI @dalerks , for "alloc eb fail" issue, please refer to: https://github.com/espressif/esp-idf/issues/3044

In WiFi driver, the TX management frames, such as beacon, probe request, probe response etc, are categorized to following type:

- If the frame len < 64, it's short management buffer. The maximum allowed short management buffer number is 8. All short management buffer are dynamic buffer.
- Otherwise, it's long management buffer. The long management buffer share the same buffer as TX data buffer which is configured via menuconfig: "menuconfig => Components => WiFi => Max number of WiFi dynamic TX buffers" if the TX buffer is dynamic or "menuconfig => Components => WiFi => Max number of WiFi static TX buffers" if the TX buffer is static.

Generally, the "alloc eb fail" error has following reasons:

- Out of memory. The reason of "alloc eb failed" in your log is out of memory because the heap only has 88 Bytes. Generally it's because the application allocates too much memory, or there is memory leak in the system, or the menuconfig is not reasonable etc.
- The second reason is because the TX malloc speed is faster than WiFi sending speed and the 8 short management buffer already allocated. Generally this may happen in AP + STA mode, especially when the channel is very busy and there are too many WiFi station is probing the AP, it's not a problem if the system can recovery soon.  In order to mitigate this case, we already make the short management buffer configurable via menuconfig, refer to https://github.com/espressif/esp-idf/issues/3044
- The 3rd reason is similar as the 2nd one, the different is it impacts the normal function, such as WiFi connection etc, and can't be recoverable. It is a bug, we need to debug it.

Alvin1Zhang, 2019-06-30:
@jasslo Thanks for reporting the issue, feel free to reopen the issue if it still exists. Thanks.

closed 2019-06-30 by Alvin1Zhang

hojatm61, 2020-02-02:
I also had this problem. I solved it with the following method.
I fixed a five millisecond delay inside the main loop of the program.
void loop () {();
...
   delay (5);
}

ZweiEuro, 2020-07-29:
I am getting this issue occasionally, as i am reading this it seems it only happens when flashing a lot and then only occasionally.
for me power cycling does the trick but i am not 100% sure

liuzfesp, 2020-07-30:
Hi @ZweiEuro, could you provide following info:

- whole log (better provide with attachment)
- IDF commit ID

ZweiEuro, 2020-08-02:
@liuzfesp  I opened a new issue. Mainly because the version is completely different.
https://github.com/espressif/esp-idf/issues/5674

duanshuai007, 2022-02-23:
我将idf版本从4.2升级到4.3后解决了这个问题。