espressif/esp-idf #1022: Data corruption when using multiple UARTs

webworxshop, 2017-09-21:
Hi,

Whilst implementing communication between an ESP32 device and two other devices via UART we have noticed data corruption after a few hundred packets are received/sent. From the point of view of the receiving UART it looks like it receives old data (i.e. the buffer read pointer is corrupted and points to old data). This corruption only occurs when two (or presumably more) UARTs are open, when sending and receiving on one, but not servicing incoming data on the other. I have uploaded an example project which reproduces the problem here: https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/uart.html#application-example

## Steps to reproduce

1. Download build and flash the project linked above, run `make monitor` to watch the output
2. Connect two USB serial adapters to the UART 1 and 2 pins specified at the top of the code
3. On the USB serial device corresponding to `MY_UART` in the code, run the provided `./serial_send.py` file.
4. On the other USB serial device send some data every few seconds
5. Wait until the program stops and logs an error, in my last run it managed to receive 420 packets and printed the following as the last few line of the log:

```
I (10809) UartTask: a1 01 00 00 00 01 02 03 04 05 06 
I (10829) UartTask: RX bytes = 11
I (10829) UartTask: Data:
I (10829) UartTask: a2 01 00 00 00 01 02 03 04 05 06 
I (10839) UartTask: RX bytes = 11
I (10839) UartTask: Data:
I (10839) UartTask: a3 01 00 00 00 01 02 03 04 05 06 
I (10859) UartTask: RX bytes = 11
I (10859) UartTask: Data:
I (10859) UartTask: 06 a3 01 00 00 00 01 02 03 04 05 
E (10859) UartTask: Count incorrect! 107270 != 420
```

6. To see the issue disappear when not listening on the second UART, comment lines 42-53 (inclusive) in `serial_basic_example.c` and re-run the test. It should run pretty much indefinitely.

## Other info

 * Seems to be somewhat related to the buffer size of the first UART, in that more packets are able to be transferred successfully when using larger buffers (may correspond to number of ringbuffer overflows?).

Any help would be appreciated.

Thanks in advance.

webworxshop, 2017-09-28:
Hi, any update on this? This is a blocker on our project.

koobest, 2017-11-08:
Hi, webworxshop:
    I have reproduce the problem with your test code. it cause by hardware fifo. The software fix is in review and will be meraged soon.

koobest, 2017-11-14:
Hi, webworxshop,
The idf has released a new version and solved the problem caused by hardware fifo. please update your idf.

closed 2017-11-21 by FayeY