espressif/esp-idf #1023: Use bluedroid from Google instead the original one?

fanthos, 2017-09-21:
Google using a patched(improved?) version of bluedroid in Android device.
The version from Google has more complete AVRC and other stack.
https://android.googlesource.com/platform/system/bt/
[Mirror](https://github.com/AospExtended/platform_system_bt)

fanthos, 2017-09-21:
From LineageOS(CynagonMod)
New: https://github.com/LineageOS/android_system_bt
Old: https://github.com/LineageOS/android_external_bluetooth_bluedroid

fanthos, 2017-09-21:
Seems that complete support in https://github.com/LineageOS/android_system_bt/tree/lineage-15.0 using `.cc` instead of `.c`.

Yulong-espressif, 2017-10-11:
@fanthos Thanks for your suggestions, we have change the bluedroid stack code to the 7.1.2 version now.

closed 2017-10-16 by FayeY