espressif/esp-idf #1025: Unable to set up WPA2_Enterprise example

RfAkramKarimi, 2017-09-21:
Hello everyone,
I am pretty new to ESP32 and have been able to do some basic stuff with this e.g. setting up MQTT example.
I am now trying to connect to a WAP2 Enterprise (PEAP) using WPA2_enterprise example but unable to do so. I have tried many suggestions made by people facing similar issues e.g.

- I have used esp_wifi_sta_wpa2_ent_clear_ca_cert();
- make erase_flash

to remove the CA certificate.

My organization does not need CA certificate as can be confirmed by the screenshot attached from my mobile phone in which I am connected to same WiFi.

I am using EAP method=1 (PEAP) in menuconfig. I have also tried generating the certificates and keys using the openssl commands given in the instructions (Readme.md) but of no use.

I have tried commenting/uncommenting the following CA certificate related APIs, same problem:
esp_wifi_sta_wpa2_ent_set_ca_cert();
esp_wifi_sta_wpa2_ent_set_cert_key();

Here is the error log:
I (306) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (316) example: 
 AK: WPA2: Enabled Successfully

I (346) phy: phy_version: 359.0, e79c19d, Aug 31 2017, 17:06:07, 1, 0
I (346) wifi: mode : sta (24:0a:c4:81:77:1c)
I (466) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1446) wifi: state: init -> auth (b0)
I (1446) wifi: state: auth -> assoc (0)
I (1446) wifi: state: assoc -> run (10)
I (1456) wpa: wpa2_task prio:2, stack:6656

Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x4011fa6d  PS      : 0x00060130  A0      : 0x801207a8  A1      : 0x3ffc7610  
0x4011fa6d: mp_read_unsigned_bin at /Users/karimim/esp/esp-idf/components/wpa_supplicant/src/crypto/libtommath.h:990

A2      : 0x3ffc93bc  A3      : 0x00000001  A4      : 0x3ffcb7e4  A5      : 0x3ffcb7e3  
A6      : 0x00000002  A7      : 0x00000005  A8      : 0x8011fa6b  A9      : 0x3ffc75f0  
A10     : 0x00000000  A11     : 0x00000008  A12     : 0x3ffc93bc  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000010  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4011fa6d:0x3ffc7610 0x401207a5:0x3ffc7630 0x4011e25a:0x3ffc7650 0x4011cddd:0x3ffc7680 0x4011cfc7:0x3ffc7730 0x40119faa:0x3ffc7800 0x4011a36a:0x3ffc7840 0x40119671:0x3ffc78d0 0x40119120:0x3ffc7920 0x40119179:0x3ffc7960 0x40117926:0x3ffc7980 0x4011712e:0x3ffc79c0 0x40115955:0x3ffc7a20 0x40115c69:0x3ffc7a60
0x4011fa6d: mp_read_unsigned_bin at /Users/karimim/esp/esp-idf/components/wpa_supplicant/src/crypto/libtommath.h:990

0x401207a5: bignum_set_unsigned_bin at /Users/karimim/esp/esp-idf/components/wpa_supplicant/src/crypto/bignum.c:114

0x4011e25a: crypto_mod_exp at /Users/karimim/esp/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal-modexp.c:40

0x4011cddd: x509_certificate_check_signature at ??:?

0x4011cfc7: x509_certificate_chain_validate at ??:?

0x40119faa: tlsv1_client_set_cred at ??:?

0x4011a36a: tlsv1_client_process_handshake at ??:?

0x40119671: tlsv1_client_handshake at ??:?

0x40119120: tls_connection_handshake2 at ??:?

0x40119179: tls_connection_handshake at ??:?

0x40117926: eap_peer_tls_process_helper at ??:?

0x4011712e: wifi_sta_get_enterprise_disable_time_check at ??:?

0x40115955: eap_sm_process_request at ??:?

0x40115c69: wpa2Task at ??:?

Rebooting...
ets Jun  8 2016 00:22:57

Is there anyone to help me in this matter?
Will be really grateful.

Thanks

XinDeng11, 2017-10-09:
@RfAkramKarimi Can you tell me the idf version you used so that I can try to reproduce

jmorsap, 2017-10-19:
I am having the exact same problem. I updated esp-idf and the toolchain a couple weeks ago.  See below.

$ git log
commit de750e99214a51c1bcfdb42b5ada931652a3c531 (HEAD -> master, origin/master, origin/HEAD)
Merge: 08b756b6 5795ccc8
Author: Angus Gratton <angus@espressif.com>
Date:   Fri Oct 6 13:14:51 2017 +0800

    Merge branch 'bugfix/pthread_join_hangup' into 'master'

    Bugfix/pthread join hangup

    See merge request !1346

XinDeng11, 2017-10-23:
@RfAkramKarimi @jmorsap We have do some system changes before, maybe you can use the lastest version try again, it is fine now. If it still have problem, please tell me your test steps and the router used, thanks.

fleaplus, 2017-10-30:
@XinDeng11 I work with @jmorsap and am helping him to get the wpa2-enterprise issue resolved.

We are also using the WPA2_enterprise example app, authenticating via PEAP. I have slightly tweaked the example code to not install the root CA or a client cert / key for the TLS method. The code with these tweaks is located here: https://gist.github.com/fleaplus/db0544c04eb12b882ba52b0b80d70b45

Using the latest master (and updated submodules) the issue still occurs. We are connecting to the radius server via an Aruba access point. I am not sure what model it is. Technically we are not able to connect via 2.1 using PEAP as the authentication method either. However we have been able to connect via TLS on 2.1, but the same configuration crashes on 3.0-dev.

XinDeng11, 2017-11-03:
Hi @fleaplus @jmorsap @RfAkramKarimi, I tried it again, it is fine on both PEAP and TLS method, I wonder have you updated the server certificates on the router? we have put the server certificates on the main folder in the example, you can try again after replace the server.pem and server.key in router server, those two certificates named wpa_server.pem and wap_server.key in the main folder, after replace them, you should rename them as server.pem and server.key.

fleaplus, 2017-11-07:
@XinDeng11 After further testing I have successfully been able to authenticate to the radius server with the provided certificate in the examples. I then tried to determine what the difference was between the ones that are provided, and what is being used by the production freeradius server we are trying to authenticate against.

It appears that the issue is the production certificates are 2048-bit, instead of 1024-bit. On esp-idf v2.1 2048-bit and 1024-bit both authenticate successfully, on esp-idf v3.0 (master) 1024-bit succeeds but 2048-bit fails with the crash above. It may be something as simple as the stack size needs to be increased for the wpa2_task, but I do not see any way to configure that in esp-idf.

If you rerun the key generation to use 2048-bit RSA, (i.e. ```openssl req -x509 -newkey rsa:2048 -keyout wpa2_ca.key -out wpa2_ca.pem```) Then install those in both the server and the example app, it should give similar results for you.

jmorsap, 2017-11-15:
@FayeY , has this issue been resolved in an update? Have you been able to recreate the error? Or will we still have issues when we need to use 2048 or 4096-bit certificates? I'm not sure why this issue has been closed. Please let us know if you need any additional information about how to recreate this error.

XinDeng11, 2017-11-16:
Hi, @fleaplus , @jmorsap , the certificate we provided is 1024 bit, NOT 2048 or 4096 bit. If you are using your own 2048 or 4096 bit certificates, please notice that the client certificate should be corresponding to your 2048 or 4096 bit certificates too. In other word, you should make sure the certificates matched on server and client, We update the certificate because it overdue,  maybe you can create the certificate by yourself if you want to use 2048 bit certificate, but please make sure the certificates on client and server matched.  If your problem is still unsolved, maybe you can provide your certificates to us for debugging

frischj, 2017-11-28:
Hi,

I still have the same problem. I try to connect to eduroam (wpa2 enterprise) with my ESP32. My esp-idf is on the latest version available (a83fbd336416c707ce3aecd20458621cac46f2c6).
I replaced the root certificate with the Telekom Root CA2 which is necessary for my connection. This seems to be a 2048 bit certificate, so I don't know, if it fits with the above problem description.
I run a slightly modified version of the wpa2_enterprise example (see code attached [wpa2_enterprise_main.c.txt](https://github.com/espressif/esp-idf/files/1509827/wpa2_enterprise_main.c.txt) ) and the corresponding certificate copied to the wpa2_ca.pem file ([wpa2_ca.pem.txt](https://github.com/espressif/esp-idf/files/1509828/wpa2_ca.pem.txt) )
My esp32 is rebooting all the time reporting this error stack [dump.txt](https://github.com/espressif/esp-idf/files/1509830/dump.txt)

It seems that the main error arrises in the certificate verification. I am quite sure, I have the right certificate and credentials (as my phone uses exactly the same settings and can connect)

0x40129ad9: bignum_set_unsigned_bin at ??:?

0x4012af9e: crypto_mod_exp at /home/frisch/esp/esp-idf/components/wpa_supplicant/src/crypto/crypto_internal-modexp.c:40

0x400fd7dd: x509_certificate_check_signature at ??:?

0x400fd9d7: x509_certificate_chain_validate at ??:?

Any ideas? Thanks already a lot!

XinDeng11, 2017-12-03:
@frischj I can not give any comments now, maybe you can attach the certificates you used and I will try to reproduce it, is that OK?

frischj, 2017-12-04:
@XinDeng11 Thanks. The certificate I used was this CA root certificate [wpa2_ca.pem](https://github.com/espressif/esp-idf/files/1526133/wpa2_ca.pem.txt). I do not have access to server certificates and keys, as the access points are managed by a different university entity. I know that this is not optimal for testing.

Hence, I set up a WPA2 radius authenticated network on my own, creating an own certificate chain, resembling the setup we have as best as I could. You can find the complete package for testing under [wpa2_enterprise_testradius.zip](https://github.com/espressif/esp-idf/files/1526202/wpa2_enterprise_testradius.zip)
I attached the esp-idf monitor dump as well as the communication wireshark listened to on the interface.

In my test case, I also tried without CA certificate on both networks (own one and eduroam), as I can be sure, that my access points here are not rogue. Even then, I get the same error. It seems, the system tries to check a certificate even if I didn't submit one. I used make erase_flash to clean up the chip before I flashed the code. I also tried to use esp_wifi_sta_wpa2_ent_clear_ca_cert() to be sure, no old certificate data is in store.

Thanks for your help. It is much appreciated.

XinDeng11, 2017-12-04:
@frischj Thanks, if it could, hope you can offer the elf file in the build folder, we need analysis the crash with the elf file, or the whole build folder. Thanks

XinDeng11, 2017-12-04:
@jmorsap @fleaplus Hi, I tried with key length as 1024, 2048 and 4096, found all of them are OK. The last time I fail to connect is due to I forget copy the ca.pem to the hostap server. Maybe you can check whether you have copied the entity certificates to both sides, in the sever part it needs ca.pem, server.crt, server.key and server.pem, in the client part it needs ca.pem, server.crt, server.key and server.pem, if there still any unclearly, please let me know.

XinDeng11, 2017-12-04:
@frischj Hi, I checked your main folder, found it short of wpa2_client.crt, wpa2_client.key and wpa2_client.pem, it is necessary for certificate check, you can try again after add the client.crt, client.key and client.pem to the main folder and rename them.

frischj, 2017-12-04:
@XinDeng11 Thanks for the comments. Please find attached a zip of the complete build folder, containing the elf file: 
[build.zip](https://github.com/espressif/esp-idf/files/1527341/build.zip)

Regarding your last comment, I didn't get why I would need the client certificates. As I understood this, I need certificates only if I want to establish a TLS connection. My connection parameters are PEAP/TTLS which should work with username and password only, together with a root CA. I removed the line using the client certificates in the code example as well. For the main, real access point, I don't have the server certificate itself. Only the corresponding CA which signed the server certificate at some point.

XinDeng11, 2017-12-06:
@frischj if it that, you should mark the ca_check too, just like that
ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    **//ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_ca_cert(ca_pem_start, ca_pem_bytes) );**
    ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EXAMPLE_EAP_ID, strlen(EXAMPLE_EAP_ID)) );
    Because when receive the server hello from server, client want to check it from the ca.pem. However, you don't have the match the client certificate. So mark down this line, and try again

frischj, 2017-12-06:
@XinDeng11 Thanks for the suggestion. I commented the line as suggested by you, removed the build folder, pulled the latest version of the esp-idf via git, erased the flash via make erase_flash and build the complete project again. And I get the same errors again. The build folder in the zip file is new, and I added the new output dump as well as a new wireshark listening to the output folder. 
[wpa2_enterprise_testradius.zip](https://github.com/espressif/esp-idf/files/1535944/wpa2_enterprise_testradius.zip)
But the system still crashes. Thank you for your help.

XinDeng11, 2017-12-13:
@frischj Hi, I reproduced the phenmenon with your code, found the root cause is the ca certificate's format can not be recognized, so that it found a wrong certificate's address when do key exchange
[image]
I suggest to mark the ca certificates in server's part in hostapd.conf like 
##### Integrated EAP server ###################################################

# Optionally, hostapd can be configured to use an integrated EAP server
# to process EAP authentication locally without need for an external RADIUS
# server. This functionality can be used both as a local authentication server
# for IEEE 802.1X/EAPOL and as a RADIUS server for other devices.

# Use integrated EAP server instead of external RADIUS authentication
# server. This is also needed if hostapd is configured to act as a RADIUS
# authentication server.
eap_server=1

# Path for EAP server user database
# If SQLite support is included, this can be set to "sqlite:/path/to/sqlite.db"
# to use SQLite database instead of a text file.
eap_user_file=/etc/hostapd/hostapd.eap_user

# CA certificate (PEM or DER file) for EAP-TLS/PEAP/TTLS
# **#ca_cert=/etc/hostapd/cert/ca-chain.cert.pem**

# Server certificate (PEM or DER file) for EAP-TLS/PEAP/TTLS
server_cert=/etc/hostapd/cert/rad_server.cert.pem

# Private key matching with the server certificate for EAP-TLS/PEAP/TTLS
# This may point to the same file as server_cert if both certificate and key
# are included in a single file. PKCS#12 (PFX) file (.p12/.pfx) can also be
# used by commenting out server_cert and specifying the PFX file as the
# private_key.
private_key=/etc/hostapd/cert/rad_server.key.pem

# Passphrase for private key
#private_key_passwd=secret passphrase  

After that, it will success and connect OK.

XinDeng11, 2017-12-13:
@frischj mark ca_cert=/etc/hostapd/cert/ca-chain.cert.pem, that's the root certificate

closed 2017-12-21 by FayeY

weisen2021, 2022-09-21:
@XinDeng11   您好，我在调试EAP-TLS时链接不上radius服务器。我使用的是idf4.4，例程是wpa2_enterprise，芯片是ESP32-D0WDQ6-V3  。
我将例程中的wpa2_ca.pem、wpa2_client.crt、wpa2_client.ket、wpa2_client.pem都换成了radius服务器中的文件。
我不用CA认证，因此我在configuraction Editor中禁止了、EAP method选择的是TLS。
在用华为手机连接radius时，会有一个用户证书选择，这个用户证书是搭建radius的同事给我的一个client.p12文件，我用手机安装这个文件即可。
但是我不明白ESP32怎么用这个client.p12文件。

weisen2021, 2022-09-21:
@XinDeng11 
Hello, I cannot connect to the RADIUS server while debugging EAP-TLS. I am using IDF4.4, the routine is WPA2_Enterprise and the chip is ESP32-D0WDQ6-V3.

I replaced wpa2_ca.pem, wpa2_client.crt, wpa2_client.ket, and wpa2_client.pem in my routine with files in the RADIUS server.

I don't use CA authentication, so I disabled it in the ConfigurAction Editor and selected TLS for the EAP Method.

When I use a Huawei mobile phone to connect to RADIUS, a user certificate is selected. The user certificate is a client. p12 file provided by my colleagues who set up RADIUS.

But I don't understand how ESP32 uses the client.p12 file.

kapilkedawat, 2022-09-21:
Please use following commands to create client certificates and keys and then replace wpa2_client.crt and wpa2_client.key with the newly generated certs and keys from command.

openssl pkcs12 -in client.p12 -out wpa2_client.crt -clcerts -nokeys
openssl pkcs12 -in client.p12 -out wpa2_client.key -nocerts -nodes

weisen2021, 2022-09-21:
@kapilkedawat 
hello, I try do that,  But I don't understand what the password is.
C:\Users\liuch\Desktop\wifi_enterprise> openssl pkcs12 -in client.p12 -out wpa2_client.crt -clcerts -nokeys
Enter Import Password:
What is this code?

kapilkedawat, 2022-09-21:
@weisen2021 password must have been shared by your colleague who gave you this file. Do you give any password in phone while installing this cert?

weisen2021, 2022-09-21:
@kapilkedawat  
After entering the password, the following information is displayed
C:\Users\liuch\Desktop\wifi_enterprise>openssl pkcs12 -in client.p12 -out client.crt -clcerts -nokeys
Enter Import Password:
Error outputting keys and certificates
3C8F0000:error:0308010C:digital envelope routines:inner_evp_generic_fetch:unsupported:crypto\evp\evp_fetch.c:349:Global default library context, Algorithm (RC2-40-CBC : 0), Properties ()

kapilkedawat, 2022-09-21:
@weisen2021 
This looks like your systems error, can you try this on some linux system? 
Or is it possible you to share the file and password somehow?

weisen2021, 2022-09-21:
@kapilkedawat  
ok, I'll try it on Linux

weisen2021, 2022-09-21:
@kapilkedawat  
Hello, I'm connected.
If I need CA authentication, what should I do?
I have copied the ca.pem and ca.key files from the RADIUS server.

kapilkedawat, 2022-09-21:
Please replace wpa_ca.pem with your ca.pem, and please select https://github.com/espressif/esp-idf/blob/master/examples/wifi/wifi_enterprise/main/Kconfig.projbuild#L25 in your example.

weisen2021, 2022-09-21:
@kapilkedawat 
hello,I replaced the ca.pem file and enabled the Validate Server through the ConfigurAction Editor. When logging in there is a domain name is required to fill in, what does this domain name refer to?

kapilkedawat, 2022-09-21:
@weisen2021 this question should be redirected to your network administrator. WiFi doesn't control this.

weisen2021, 2022-09-22:
@kapilkedawat 
Thank you very much for your help. I have successfully connected to the RADIUS server with CA certificate.