espressif/esp-idf #1026: Feature Request: Command line specify make flash com port

lucazader, 2017-09-21:
Hi all,

I am often having to switch between different dev boards and flashing them with the same or similar code.
Whenever i do this i have to change the com port in the menu config to `make flash` the target.

This is fine, apart from that this triggers a full re-build of the code base taking a significant amount of time.

Can we have the option in the `make flash` command to specify the com port to flash to?
Eg `make flash` would use the default com port that is set in the menuconfig (eg com4)
`make flash --port=com6` would instead flash to com6 instead, without causing a code re-build

Would be good if this could be added to `make monitor` as well.

Thanks all!

projectgus, 2017-09-21:
You can actually do this now, try `make flash ESPPORT=COM6`. You can also set ESPBAUD (I have that one set in my .profile)

This is clearly not documented well enough though, so please leave this issue open and we'll figure out how we can make it clearer.

lucazader, 2017-09-21:
Fantastic!

Yea I couldn't seem to find any doccumentation around this. 
If it helps I was mostly looking at this page http://esp-idf.readthedocs.io/en/latest/get-started/make-project.html
Just in case the place I was looking was not the place that i was expected to look.

projectgus, 2017-09-21:
It looks like right now it's only mentioned in the menuconfig item help for port & baud rate:
http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#esptoolpy-port

Putting it on the page you linked as a "tip" seems like a good suggestion to me.

lucazader, 2017-09-22:
Great!

Thanks for the quick responses!

krzychb, 2017-09-22:
I used to strugglle with the same issue like @lucazader as well. 

This applies more to Windows users because of assignment of different COMn for different broads depending on the type of USB to serial chip. On Linux the port is usually just `/dev/ttyUSB0` (or `/dev/ttyUSB1` for ESP-WROVER-KIT).

 I propose to add information about both `ESPPORT` and `ESPBAUD` in http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html as well.

closed 2017-12-21 by FayeY

pglen, 2018-12-17:
Great! This helped me immensely.  This way I can program both the client side of my project and the server side without plugging them in / out. THANKS>