espressif/esp-idf #1027: SPI slave communication not working on latest master code

gcarstoiu-cr, 2017-09-22:
Hello all,

I'm working on a project which uses the ESP32 as a SPI slave to communicate with an application MCU.
Until recently, I've been using esp-idf release v2.1, but I decided to use the latest code from the master branch.
After doing the update, the SPI code did not work anymore. 
I've been looking at the differences btw. v2.1 and master, and I saw that the SPI master code has been updated with a new method for handling DMA, but the slave driver has not.
This was done in this commit: https://github.com/espressif/esp-idf/commit/b834fcf78aa4cc16d4c3f5349286df3580d24edb

I've forked the esp-idf repo and uploaded the fix to a branch. I will do a pull request shortly.
Here is the commit that I've made:
https://github.com/gcarstoiu-cr/esp-idf/commit/34326b744b1e94033ef9f3e042ee26583df9d6a5

Also, do you think that other improvements to the SPI master code should be ported to SPI slave code?

I'm asking because I've encountered another issue, in which the received bytes sometimes get corrupted, and I saw that this only happens while simultaneously transmitting data over WiFi using the standard socket send() function.
Has anybody else encountered this? I will raise a new issue for it, I think.

Thank you!
Gabriel C.

ginkgm, 2017-09-30:
thanks for your reporting, we'll fix this. 
TW#15670

ouki-wang, 2017-10-13:
I meet this problem too.
today, I tried 7 hours before I find this issue.
How long we can go to right?keep seeing.

gcarstoiu-cr, 2017-10-16:
@ouki-wang , please see my original post, where I said that I fixed the issue and uploaded the fix on a branch that I have created after for king the public esp-isf repo. You can merge that branch into your code and it should work.

ouki-wang, 2017-10-17:
@gcarstoiu-cr , I am sorry but have to say that the code is still wrong. The slave could not receive the final data. The return data of the slave  is unstable and always go wrong.

closed 2017-10-18 by igrr

ginkgm, 2017-10-18:
hi @gcarstoiu-cr , we've updated the IDF to solve this issue. please update to the latest and see.

@ouki-wang , please also update, and check

1. if the speed is too high. To see this, use 1MHz or lower
    if the data get's correct, maybe it's a timing issue, please report and I'll solve it.
2. if this is caused by DMA. To see this, initialize the slave with DMA channel 0
    for DMA issue, there're mainly two causes:
    1. due to silicon issue, the transaction length should be multiples of 4 bytes, or the last data will not be written to buffer by DMA.
    2. if the transaction length is multiples of 4 bytes, but the data is still last, maybe this is from the driver issue, I'm working on this issue now.
3. if this is caused by the early CS-inactive issue in ESP master. To see this, add cs_ena_posttrans=5 to the master initial parameter.

L-eeX, 2017-10-19:
@ginkgm 
**master：**
 Received: Sender, transmiゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ
Received: Sender, transmiゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ
Received: Sender, transmiゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ
Received: Sender, transmiゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ
Received: Sender, transmiゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥゥ
**slave：**
Received: This is the receiver, sending data for transmission number 2313.
Received: This is the receiver, sending data for transmission number 2314.
Received: ^豸頃€
Received: ^曯頃€
Received: ^枍%褉
Received: ^枍%褉
Received: ^杹畞€
Received: This is the receiver, sending data for transmission number&FfF?
Received: This is the receiver, sending data for transmission number 2321.

sankarsurya, 2018-04-30:
is there having any sample code for esp32 spi slave

sankarsurya, 2018-04-30:
i having a project on spi communication between stm32 and esp32 so stm32 was a master and esp32 was a slave am using is there having any esp32 slave code

ginkgm, 2018-05-02:
@DF-LeeX Could you please open a new issue and describe this more detailedly?

ginkgm, 2018-05-02:
@sankarsurya Yes, it's in example/peripherals/spi_slave folder, the slave is ``receiver``.