espressif/esp-idf #1030: build error when 2 server certificates of size 5k and 5.5k is used with https

ikawaljeet, 2017-09-23:
I am getting following build error when i use 2 server certificates since i need to send request to 2 different server:
-------------------------------------------------------------------------------------------------------------------------------
../project/build/main/libmain.a(firmware_status_update.o):(.literal.fw_update_task+0x4): undefined reference to `_binary_server1_pem_end'
../project/build/main/libmain.a(firmware_status_update.o):(.literal.fw_update_task+0x8): undefined reference to `_binary_server1_pem_start'
collect2: error: ld returned 1 exit status
----------------------------------------------------------------------------------------------------------------------------------
When i use one certificate it works fine.

output of make partition_table:
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,1536K,
ota_1,app,ota_1,0x190000,1536K,
*******************************************************************************
and make size output:
Total sizes:
 DRAM .data size:   14752 bytes
 DRAM .bss  size:   57304 bytes
Used static DRAM:   72056 bytes ( 190088 available, 27.5% used)
Used static IRAM:   75028 bytes (  56044 available, 57.2% used)
      Flash code:  832003 bytes
    Flash rodata:  289396 bytes
Total image size:~1211179 bytes (.bin may be padded larger)

size of server1.pem is 5K
size of server2.pem is 5.5K

I have read all binary embedded goes to .rodata section of flash, but i don't know what is happening at memory level.

Please help.

negativekelvin, 2017-09-23:
What is your component.mk look like?

ikawaljeet, 2017-09-23:
Components.mk file:
***************************************************************************************************************
\#
\#  "main" pseudo-component makefile.
\#
\# (Uses default behaviour of compiling all source files in directory, adding 'include' to include path.)

\# embed files from the "certs" directory as binary data symbols
\# in the app
\#COMPONENT_EMBED_TXTFILES := wpa2_ca.pem
\#COMPONENT_EMBED_TXTFILES += wpa2_client.crt
\#COMPONENT_EMBED_TXTFILES += wpa2_client.key

COMPONENT_EMBED_TXTFILES := server1.pem
COMPONENT_EMBED_TXTFILES := server2.pem

jkearins, 2017-09-23:
Seems like you missed '+':
`COMPONENT_EMBED_TXTFILES += server2.pem`

ikawaljeet, 2017-09-23:
Thanks jkearins,
I was missing that only, i dont know how i can miss that, this was there in commented certificates. It's building now.  :)

closed 2017-09-23 by ikawaljeet