espressif/esp-idf #1032: Errors with latest toolchain ...

nkolban, 2017-09-23:
I am building apps using ESP-IDF master which requires a new tool chain.  See here for details of the toolchain instructions:

http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html

I have installed and am building with:

https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-73-ge28a011-5.2.0.tar.gz

My code compiles cleanly ... however, when I try and run C++ code that previously worked, we now hang.  Here is the code I am executing:

```
void task_test_sandbox(void *ignore) {
	ESP_LOGD(LOG_TAG, "About to create a string stream");
	std::stringstream ss; /// <---- This is where we hang
	ESP_LOGD(LOG_TAG, "About to write to string stream");
	ss << "Hello World!";
```
The code is trivial ... and we appear to hang when creating an instance of `std::stringstream`.  The console log writes:

```
D (1379) task_test_sandbox: About to create a string stream
V (1385) esp_pthread: pthread_once: call init_routine 0x3ffc0940
```

As we see, we log the first diagnostic statement and then there is a new internal `esp_pthread` message but after that ... we don't appear to return and just hang.

My environment is latest toolchain and latest ESP-IDF as of the date of this post.

chegewara, 2017-09-28:
On arduino and with latest toolchain ive got this message:

`E (6066) BT: Device already in Resolving list`
`E (6269) esp_pthread: pthread_getspecific: not supported!`
`E (6270) esp_pthread: pthread_setspecific: not supported!`
`abort() was called at PC 0x400e92cf on core 0`
and long:
`Backtrace: 0x400893a8:0x3ffe2360 0x400894a7:0x3ffe2380 .....`

nkolban, 2017-09-28:
I heard from an Espressif rep today that they had recreated the issue using a sample I had supplied and had thoughts on how it might be resolved.  We should continue to be patient as it might be a while before a proper resolution will be forthcoming.  There are some public holidays coming up over the next week or so that may delay normal resolution of issues.

nkolban, 2017-11-19:
Resolved.

closed 2017-11-19 by nkolban