espressif/esp-idf #1034: uart_read_bytes and uart_write_bytes has different buffer data type

fanthos, 2017-09-24:
These functions using different type for buffer.
The read function using `unsigned char *` and write function using `char *`.

`int uart_read_bytes(uart_port_t uart_num, uint8_t* buf, uint32_t length, TickType_t ticks_to_wait);`
`int uart_write_bytes(uart_port_t uart_num, const char* src, size_t size);`

FayeY, 2017-11-28:
Hi @fanthos , thanks for your reporting! This issue should have been solved in the latest IDF. If it is not, please feel free to reopen this issue. Thanks.

closed 2017-11-28 by FayeY

Quentinb, 2019-12-20:
It looks like this is still the case? Checked the code in the master branch and release 4 and both are still different types.
[image]
[image]