espressif/esp-idf #1036: RMT in loop mode adds extra idle cycle to output

DavidAntliff, 2017-09-25:
I'm using RMT to generate a fixed frequency pulse wave.

```
void rmt_task(void * pvParameter)
{
    rmt_config_t rmt_tx = {
        .rmt_mode = RMT_MODE_TX,
        .channel = RMT_CHANNEL_0,
        .gpio_num = GPIO_NUM_4,
        .mem_block_num = 1,
        .clk_div = 80,  // 1 MHz
        .tx_config.loop_en = true,
        .tx_config.carrier_en = false,
        .tx_config.idle_level = RMT_IDLE_LEVEL_LOW,
        .tx_config.idle_output_en = true,
    };
    rmt_config(&rmt_tx);
    rmt_driver_install(rmt_tx.channel, 0, 0);

    size_t num_items = 4;
    rmt_item32_t * items = calloc(num_items, sizeof(*items));
    for (size_t i = 0; i < num_items; ++i)
    {
        items[i].level0 = 1;
        items[i].duration0 = 1;
        items[i].level1 = 0;
        items[i].duration1 = 1;
    }

    rmt_write_items(RMT_CHANNEL_0, items, num_items, false);

    while(1)
        ;
}
```

I was expecting this to generate a regular 500kHz high/low pulse train, with each high pulse 1us wide, and each low pulse also 1us wide. However at the end of the loop, there is an extra cycle added at the "idle" level:

[image]

If I increase the `duration0/duration1` fields to, say, 4, then it's obvious that the extra idle time is equal to clk_period * clk_div. I.e. 1us in this case:

[image]

I can accommodate this by shortening the final low pulse by 1 cycle.

```
    // post-loop:
    --items[num_items - 1].duration1;
```

Is this intended behaviour?

costaud, 2017-10-19:
This seems to be a hardware issue, we can only workaround it for now. To generate a clock, you can also use LEDC module.

FayeY, 2018-01-05:
Hi, sorry for the inconvenience, we will update the document for it.

closed 2018-01-05 by FayeY