espressif/esp-idf #1037: WiFi Promiscuous filter breaks w/ WIFI_PROMIS_FILTER_MASK_ALL

SoftwareSamurai, 2017-09-25:
After calling esp_wifi_set_promiscuous_filter() with the filter set to WIFI_PROMIS_FILTER_MASK_ALL and starting up the WiFi, I randomly get this output:

"E (51478) wifi: wdev.c wdevProcessRxSucDataAll 1008"

When I get that, the Promiscuous callback appears to never get called. But it doesn't happen all the time. In fact, I can reset the ESP32 via the button (not even re-uploading my program) and sometimes it happens and sometimes it runs fine.

--Details--
Board: ESP32 Dev Module, Rev 0.
IDE: PlatformIO (Also tried this on Arduino IDE - same results)
Core: Latest arduino-esp32 branch.

(Note: Originally reported on the arduino-esp32 repro, but was asked to move it over here.)

liuzfesp, 2017-10-09:
HI @SoftwareSamurai I didn't see this issue before, but I think this is a bug in WiFi, will update you when we find the root cause

malaimoo, 2017-12-15:
esp_wifi_set_promiscuous_filter is ok  only  when WIFI_PROMIS_FILTER_MASK_MGMT

TimXia, 2018-01-24:
@SoftwareSamurai @malaimoo The issue has been solved in IDFv3.0. Please pull the latest IDFv3.0 to test. Thanks,

TimXia, 2018-01-26:
@SoftwareSamurai @malaimoo The IDF commitID 891c1f4a2b67334dac2d38ce9787069f0ab5fbb4 and cdd7a876742e4be73fe9798095360ae1856bf849 have solved this issue. Please have a try.

closed 2018-02-24 by FayeY